gnuton/asuswrt-merlin.ng #402: TUF-AX3000 Router recovery factory issue

qqw1231, 2023-05-29:
388.2.2Factory RecoveryWhen filling out a telecommunications network account, it will pop up a new broadband address again, because when I fill in the first half, a new broadband address will pop up again, and the original one is still filled in

qqw1231, 2023-05-29:
It should be said that when I fill in the router's WIFI account password, it will pop up a new header asking me to enter the telecommunications account password and start filling it out again. The one I previously filled in is still there

gnuton, 2023-05-29:
can you please tell me which version you tested? the TUF or NO_TUF image?
Did you factory reset after flashing the new firmware from stock?

qqw1231, 2023-05-30:
> 

It should be that I have been asking some questions since 388.1, without the original UI of Tuf

qqw1231, 2023-05-30:
I am using the 388.2.2 version of the UI that you have modified, and the official version also has this issue

qqw1231, 2023-05-30:
Asus official firmware also has this issue

gnuton, 2023-05-30:
I am not sure I completely got the issue although.
In order to understand it I would need a step to step intructions on how to reproduce them.
Please change also the title to something more descriptive