gnuton/asuswrt-merlin.ng #404: DNSFilter.asp in TUF UI menus

gnuton, 2023-05-31:
**Router Model Affected**
Models:  TUF-AX3000 with tuf ui

**Firmware Version Affected**
388.2

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
http://router.asus.com/DNSFilter.asp is missing.

**Expected behavior**
remove the page from the menu

encip, 2023-06-17:
VPN Director tab is missing also in TUF UI.

gnuton, 2023-06-19:
@HunHawk it's not about updating the filename, it's just something I need to update in the menu files.
In any case this is in the TODO list for the next release.

This bug doesn't exist for the image without TUF_UI

gnuton, 2023-09-25:
Fix 2ba2da28b43fa0f939f22d7693f289bdc2cf8765

gnuton, 2023-09-26:
Fix 1c6144ead939723e36bf52a17a92c41c3c12f7d1