gnuton/asuswrt-merlin.ng #405: Can't update main mesh router to latest 388.2_2_0 FW

dacootmeister, 2023-06-02:
**Router Model Affected**
Models: ZenWiFi_XT8 (base model: RT-AX95Q)

**Firmware Version Affected**
Version : 388.2_2_0

**Is this bug present in upstream Merlin releases too?**
IDK

**Describe the bug**
I flashed my 3 mesh nodes with the 388.2_2_0 firmware. However, the main node keeps reverting back to the latest beta. This does not apply for the slave nodes, these get upgraded to 388.2_2 nicely.
The main node is however running multiple AWM scripts. USB Disk check showed no errors.

Expected behavior
I would like the main node also to be updated to 388.2_2

The logfiles show no real reason for the flash to fail and revert back to beta 2.

May  5 07:05:22 kernel: proc_dostring_crashlogsave:  crash log filename is /jffs/crashlog.log
May  5 07:05:22 kernel: crashFileSet: log signature invalid ! 

[image]

gnuton, 2023-06-02:
Hi,
which version was running before on the nodes?

dacootmeister, 2023-06-05:
> Hi, which version was running before on the nodes?

it was RT-AX95Q_388.2_2-gnuton0_beta2_puresqubi

gnuton, 2023-06-05:
I will check if I can reproduce this, but since it worked for the node it should work for master too.
In any case you could try to backup your changed, reset to factory, upload the new firmware and then restore the backup

dacootmeister, 2023-06-05:
unfortunately, it isn't that logical in my case so I will backup my config, reset to factory and restore tomorrow (less ppl in the house)

dacootmeister, 2023-06-05:
> when you flash the device via web the flasher doesn't flash the ROM straightaway but it prepares the system for it. Maybe some scripts or something got corrupted or prevent this on your system and the flasher is not able to continue. If you are sure the firmware is right and you are BRAVE ENOUGH, you can also download the image with wget http://file and then flash it via CLA with hnd-write file.w This will flash the live system which is not ideal at all and not as safe as the web flash.

Bit risky perhaps, so I'll try option 1 first before reverting to the command prompt :-)

dacootmeister, 2023-06-06:
Removed USB drive, flashed again solved the issue.
Thanks, no bug from FW perspective

[image]
[image]