gnuton/asuswrt-merlin.ng #406: Traffic Analyzer Inoperative RT-AXE95Q Zenwifi ET8 388.2_2-gnuton1

phostetl, 2023-06-02:
**Router Model Affected**
Models: RT-AXE95Q

**Firmware Version Affected**
388.2_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
“Statistic” page doesn’t show any clients or traffic. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web interface -> Traffic Analyzer -> Statistic
2. See error

**Expected behavior**
This worked in the last beta giving a spiffy pie chart. 

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section
[image]

gnuton, 2023-06-02:
It works for me -_-
Can you try to check if it works in an incognito browser? if it doesn't you may wanna try to reset to factory
[image]

dacootmeister, 2023-06-05:
> It works for me -_- Can you try to check if it works in an incognito browser? if it doesn't you may wanna try to reset to factory [image]

it doesn't for me, even in Incognito mode:
[image]

gnuton, 2023-06-05:
Interesting. I will try to reset my router and try again.

phostetl, 2023-06-08:
I, too, have tested with an incognito browser and it still doesn't work.  I found the following.  I don't know if this was incorporated into the gnuton release.

https://www.snbforums.com/threads/traffic-analyzer-statistics-showing-no-data-asus-rt-ax86u-on-latest-stable-firmware.84789/

[Apr 29, 2023](https://www.snbforums.com/threads/traffic-analyzer-statistics-showing-no-data-asus-rt-ax86u-on-latest-stable-firmware.84789/post-838284)
[Add bookmark](https://www.snbforums.com/posts/838284/bookmark)
[#2](https://www.snbforums.com/threads/traffic-analyzer-statistics-showing-no-data-asus-rt-ax86u-on-latest-stable-firmware.84789/post-838284)
From Asuswrt 388_23012 changelog:

- Fixed the issue where Traffic Analyzer sometimes couldn’t record data.

Asuswrt-Merlin 388.2 is based on 388_22525.

phostetl, 2023-06-08:
If there was a method to export and import DHCP reservations I'd hard reset the units and start over.  If not I'll have to pass because it is too disruptive to everyone in my family.

gnuton, 2023-06-08:
@phostetl no we are using a previous GPL. The fix must come from asus, so it wil lbe fixed in the next release

gnuton, 2023-06-08:
@phostetl let me test it.

micron888, 2023-06-28:
i am having same issues on tuf-ax5400 with 388.2_2_0-gnuton1
it works for a few days, and start to show blank like above

gnuton, 2023-06-29:
We are in contact with ASUS which promised new GPLs to come for us  in the next weeks.

CPngN, 2023-06-29:
Awesome, fresh ET8 code! :)
388_2.2 has been wonderful in comparison to old stock 386.
I discovered my Parental Controls were empty but could be recreated via the
app. It tries to recreate what I set in the web UI but doens't quite get it
all - so if you use parental controls on 388, stick with the phone app!

phostetl, 2024-03-23:
No longer an issue with 3004.388.6-gnuton1.

closed 2024-03-23 by phostetl