gnuton/asuswrt-merlin.ng #407: RT-AX82U V2 support

gnuton, 2023-06-02:
This thread is about supporting RT-AX82U v2.
If you see this issue here is because I have decided to support this model.

So, feel free to ask questions here about this topic.
I will keep the status updated.

Status:
- [] received GPL
- [] received development unit
- [] Merged GPL to feature-branch
- [] Initial testing & Bugfixing phase
- [] Public pre-alphas
- [] Alpha released
- [] collecting testing reports  & Bugfixing phase
- [] Beta released
- [] collecting testing reports  & Bugfixing phase
- [] Stable released

Download firmware:  NA

Roki100, 2023-06-04:
does the hardware of v2 differ much to v1?

gnuton, 2023-06-04:
@Roki100 IIRC Asus has just updted the 5Ghz, but for merlin, it's a completely a new model/terget.

ChrisMa789, 2023-06-04:
> @Roki100 IIRC Asus has just updted the 5Ghz, but for merlin, it's a completely a new model/terget.

Long awaited, thank you for your work

bdkacz, 2023-06-23:
Any date of alpha release ? :)

gnuton, 2023-07-10:
@bdkacz no date yet. Hopefully it will come with the next release.
The next release should have alpha soon.

gnuton, 2023-07-26:
I had to get around a few issues since this is using very different SDK than the V1.
But here we have the first screenshot
[image]

bdkacz, 2023-07-26:
Great job :)

Psychoboy, 2023-07-26:
> I had to get around a few issues since this is using very different SDK than V1. But here we have the first screenshot [image]

Looking great, can't wait to use it.

huwu2705, 2023-07-26:
I'm happy to see that the development is progressing. All the best and I hope we can test a first version soon

adrijanh9, 2023-08-12:
Why does the thread for RT-AX58U V2 under "New models to support for 2023" link here?

bdkacz, 2023-09-25:
> Beta images are out

I've been testing it since it was released, so far no problems and finally QoS works as it should, because it was average with the original firmware

tjbbjt, 2023-09-30:
> Beta images are out

Do beta images work with RT-AX58U v2 or is that future development?