gnuton/asuswrt-merlin.ng #408: DSL-AC68U (386.xx) erroneously reports update available (showing 388.xx)

mackonsti, 2023-06-03:
**Router Model Affected**
Asus DSL-AC68U

**Firmware Version Affected**
386.9-gnuton1 Stable

**Is this bug present in upstream Merlin releases too?**
Not applicable

**Describe the bug**
My router has the latest Stable firmware flashed (386.9-gnuton1 from January 2023) and I got a blinking icon on home page that an update is available. However in Releases, the only new stable version is the non-compatible 388.xx version.

**To Reproduce**
Steps to reproduce the behavior:
1. Login today in the DSL-AC68U router.
2. Check that there is a flashing bell icon on top right, click on it.
3. The page displayed is "Administration - Firmware Upgrade"
4. Erroneous/incompatible firmware prompted to update:

```
Current Version : 386.9-gnuton1
Manual firmware update : Upload
The latest version : 388.2_2-gnuton1
```

**Expected behavior**
Not sure if it is possible to discriminate between 386.x and 388.x firmwares in your update mechanism, hopefully only 386.x compatible builds should be available on the Asus WRT interface.

**Screenshots**

[image]

[image]

@gnuton thank you for your continued support and updates for this generation of Asus models!  I decided to move back to your firmware after some time using the stock one, on my primary home DSL modem router.

Not sure if you are aware of this issue where 386.xx firmware prompts for updates when a 388.xx non-compatible firmware is pushed in Releases here. Just to inform you, not complain. Many thanks again for your support!

Grazie mille.

gnuton, 2023-06-03:
Thanks for reporting i need to fix the server for this.

gnuton, 2023-06-04:
Scripts updated and problem fixed! Thanks a lot!

closed 2023-06-04 by gnuton