gnuton/asuswrt-merlin.ng #409: Traffic Analyzer toggle button missing in TUF-AX5400

evolsano, 2023-06-04:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
388.2_2-gnuton1_tuf

**Is this bug present in upstream Merlin releases too?**
Not sure.

**Describe the bug**
Unable to enable traffic analyzer in the router. the toggle is missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Traffic Analyzer'
2. The toggle is missing

**Expected behavior**
Expect the toggle is there.

**Screenshots**
[image]

The toggle show in Asus stock firmware.
[image]

gnuton, 2023-06-04:
can you please test it opening the page in an incognito browser page?

evolsano, 2023-06-05:
> can you please test it opening the page in an incognito browser page?

Tested in Chrome incognito. Still not showing the toggle button.

gnuton, 2023-06-05:
I will check it better. thanks!

sirob1, 2023-06-10:
I can also confirm this issue.