gnuton/asuswrt-merlin.ng #411: WLAN SSID has already been added - egress_port 7

michel130, 2023-06-07:
**Router Model Affected**
ASUS ZenWiFi AX (XT8)

**Firmware Version Affected**
388.2_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
Unknow this was in a old version also.

**Describe the bug**

This popups in the log, for the rest the router works, no connection lost.

Jun  7 10:53:44 kernel: ^[[0;33;41m[ERROR archer] archer_mcast_activate,577: ADD_PORT: WLAN SSID has already been added: egress_port 7, current 0x0004, new 0x0004^[[0m

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'system log'
2. Click on [Default message log level] 'Notice'
3. Click on [Log only messages more urgent than] 'info'
4. See error

**Expected behavior**
Nothing happens, only that this message is poping up many times.

**Screenshots**
-

michel130, 2023-06-08:
Hi, you can close this issue.
I never had this message before so for me it was very strange.

This new software is stable for me 4 days 18 hour(s) 56 minute(s) 7 seconds

The backhaul (wifi) is stable with FIX Channel bandwidth 80 Mhz and Control Channel 100 no orange fallbacks to 2.4 network.

gnuton, 2023-06-08:
it's coming from the NAT acceleration driver, according to RMerlin it's more a debug message than an error

closed 2023-06-08 by gnuton