gnuton/asuswrt-merlin.ng #412: Unable to add DSL-AX5400 as AIMesh Node

thesuperben, 2023-06-07:
**Router Model Affected**
Models: DSL-AX82U/DSL-AX5400

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
When booting router initially after flashing firmware it shows that it is still an AX5400. Unable to use as AIMesh Node.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash firmware from stock AX5400 to 388.2_2_0-gnuton1
2. Check Tools
3. Model shows DSL-AX5400
4. Features show AX5400 features. odmpid=DSL-AX5400 when running nvram get odmpid from SSH.

**Expected behavior**
Expected flashing to change from DSL-AX5400 to DSL-AX82U. I have been able to manually resolve the issue by running an init-nat script in /jffs/scripts/ each boot from [here](https://forums.whirlpool.net.au/thread/32k1vyn9?p=19) however this means I am unable to load the device as an AIMesh Node because it factory resets to do so which loses my script.

DJE007, 2023-06-08:
This can be done on a DSL-AX82U, I have it running that way currently.

[image]

If you are talking about the DSL-AX5400 it's not supported in the hardware.

DJE007, 2023-06-08:
A work around could be don't fully reset to factory settings with;

nvram erase && nvram set model=DSL-AX82U && nvram set productid=DSL-AX82U && nvram commit &&  reboot

thesuperben, 2023-06-08:
> A work around could be don't fully reset to factory settings with;
> 
> nvram erase && nvram set model=DSL-AX82U && nvram set productid=DSL-AX82U && nvram commit &&  reboot

Unfortunately when I select to set the device as a node it immediately prompts to factory reset. Is there a way around this?