gnuton/asuswrt-merlin.ng #414: Guest network page seems broken/bugged, Zenwifi AX8, RT-AX95Q, 388.2_2_0

jazha, 2023-06-14:
**Router Model Affected**
Models: (eg: ZenWifi AX8, RT-AX95Q)

**Firmware Version Affected**
RT-AX95Q_388.2_2_0-gnuton1_puresqubi

**Is this bug present in upstream Merlin releases too?**
I'm not sure

**Describe the bug**
The page to setup Guest Network(s) seems to be broken

**To Reproduce**
1. Login to router
2. Click Guest network
3. Page Looks like the pictures attached

**Expected behavior**
I expect the page to make sense, as it is now, it looks broken and unintuitive.

**Screenshots**
[image]
[image]

gnuton, 2023-06-14:
does it look any better if you open the page in an incognito browser?

jazha, 2023-06-15:
> does it look any better if you open the page in an incognito browser?

Sorry for the late response. It does not look any different in incognito mode

Wyehoota, 2023-06-16:
Click one of the "enable" buttons, and set/accept values - see if that then looks more like you expected. In you grab, all guest networks are disabled.

jazha, 2023-06-16:
I'm away for the weekend, will check when I'm back. 

However the landing page should not look like that, the design is highly confusing and unintuitive, and I suggest it is changed so you actually know what you click enable for, and maybe remove the extra(?) buttons with enable. Having 3 buttons of each wireless seems excessive. 

Anyway, I will check back on sunday evening probably, try to click enable to see.

jazha, 2023-06-20:
I see now, there are 3 possible guest networks per wifi channel or something. 3 for 2.4Ghz, 3 for 5Ghz-1 and 3 for 5Ghz-2.

It was just not very intuitive, and it might be the way it looks in the official software too, I dont know. Anyway, clicking the buttons provide the interface I expected. Not really a bug, more UI improvement could be done to explain that button 1 is network 1 etc.

closed 2023-06-20 by jazha