gnuton/asuswrt-merlin.ng #416: Guest network DNS DHCP issue

Roki100, 2023-06-15:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.2_2_0 stable

**Is this bug present in upstream Merlin releases too?**
I dont know

**Describe the bug**
So i announce only one dns ip on my lan, its my raspberry pi pihole instance, on stock firmware every intranet isolated gusest network announced router's ip for those clients, but on merlin it still announces raspberry's ip so clients cant access the internet because they cant connect to the announced dns server

**To Reproduce**
Set single dns to lan device and make intranet isolated guest network, see if it still announces lan device clients there cant access or router's ip like its supposed to

**Expected behavior**
Announce router's ip on isolated guest networks when lan ip is set as dns

Roki100, 2023-07-09:
Hey @gnuton any input on this? maybe a temporary workaround or fix? it has given me headache already as it cripples my entire network 😔

gnuton, 2023-07-11:
HI @Roki100 guest network is a closed source component from ASUS.
I have just got the new GPLs and merged, so a pre-alpha is available if you are brave enough and ASUS may have fixed this.
https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1537/workflows/ccb6862d-a755-4398-b939-e580d0793679/jobs/4377/artifacts

Note this most of these images are untested and could brick your router (Which you can recover with TFTP) or may have sever bugs.

Roki100, 2023-07-22:
hey @gnuton i waited for a github new release, just for more safety, installed 388.4_0-gnuton0_alpha1 today and the behavior is still the same

for more context

on stock firmware:
Router IP: 192.168.1.1
DNS IP Of Pi-Hole: 192.168.1.4
Guest network isolated: The router announces 192.168.1.1 as DNS probably because it knows the clients cant access the 192.168.1.4 ip
Guest network non-isolated: The router normally announces 192.168.1.4 as DNS

on merlin:
Router IP: 192.168.1.1
DNS IP Of Pi-Hole: 192.168.1.4
Guest network isolated: The router announces 192.168.1.4 as DNS which clients cannot access due to the network being isolated
Guest network non-isolated: The router normally announces 192.168.1.4 as DNS

If that is a closed source component there must be something going on with some merlin's component then as it happens only on merlin

gnuton, 2023-07-22:
This behaviour fallback solution if really works like this sounds like a workaround to me.  
I am not sure there is so match smartness in the router rules for understanding that IP is in the LAN and it cannot be reached.  I am wondering if this was part of the design or just accidentally work like this for this case.

IMHO a proper solution would be to broadcast different DHCP settings per private network instead of the same for all, as you see now.

In any case I have not tried this setup yet. So no idea about what's really happening.