gnuton/asuswrt-merlin.ng #417: ET8 CPU Frequency Not Shown (shows ??? instead of frequency)

networknoob123, 2023-06-16:
**Router Model Affected**
Models: ET8

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
The Tools - System Information (click on Tools on the menu on the left) does accurately display the CPU frequency.  It only shows ???.  Devices are used in AiMesh mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Login to router via web GUI (I used Firefox in incognito mode)
2. From the menu on the left side click on Tools
3. On the Tools - System Information page look for the CPU section
4. Find CPU Frequency
5. See bug that shows '???' instead of actual frequency

**Expected behavior**
I would expect to see the actual frequency (or max frequency).  The CPU Load Average (1, 5, 15 mins) is shown so I assume the frequency is detected.

**Screenshots**
Screenshot attached.
[image]

gnuton, 2023-06-16:
Thanks for reporting this. I think this is a regression.

DiMENZi0N, 2023-08-30:
Hello there. This is still an issue for ET8 in 388.4 BETA3

[image]

commit 0770fd72 (2021-10-19): webui: only forbid setting protocol to http-ony if in router mode (fixes #417)