gnuton/asuswrt-merlin.ng #418: Unable to change wifi "tx power" setting

arvidjo, 2023-06-18:
**Router Model Affected**
Models: (eg: XT8)

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
Unable to change wifi "tx power" setting. It defaults to "performance" but when I change the slider I am unable to save changes. 

**To Reproduce**
Steps to reproduce the behavior:
1. Wireless settings
2. Click on Professional
3. Scroll down to 'tx power setting'
4. See error

**Expected behavior**
I expect to be able to change this setting

**Screenshots**
n/a

gnuton, 2023-06-18:
Hi,
have you tried to reset to factory after flashing the firmware?

arvidjo, 2023-06-19:
Hi, yes. I have done a factory reset after flashing. I also forgot to mention I run this router in AP-mode.

I did not have this problem with previous gnuton version though.

closed 2023-07-01 by arvidjo