gnuton/asuswrt-merlin.ng #419: VPN Director based on url and domainnames

jwh75, 2023-06-18:
**Is your feature request related to a problem? Please describe.**
I would like to use VPN only for some applications/urls e.g. NETFLIX

**Describe the solution you'd like**
VPN Director shall allow the user to select remote domainname/url

**Describe alternatives you've considered**
When thinking about this I had in mind Firewall->URL filter, and I wonder if something similar could be applied on what shall be redirected to VPN or to WAN

gnuton, 2023-06-18:
https://www.youtube.com/watch?v=oBLS7Wkn0C4&ab_channel=MoritzFl

jwh75, 2023-06-18:
Thanks a lot for your answer.

I could be wrong, but I think this video explains how to select which
devices connected to the router shall use VPN and which ones shall connect
using Wan (not VPN).

My request is different... Once a device is supposed to use VPN I would
like to select which url shall visit using VPN and which one shall be
routed directly.

I think this could be done properly setting Destination IP... But just to
give an example, how can this be done for netflix? (supposing I want VPN
only active when visiting netflix?)
Which ip shall be added in destination ip? If I could have a destination
url instead of a destination ip it would be much easier. Isn't it?

Did I make myself clear now?

Is that somehow possible?

Thanks

jwh75, 2023-06-18:
Hey, to better explain,

here there is a similar request (even if the other way round... He would
like to have everything on VPN but Netflix, I am looking for the
opposite... But the concept is the same).

https://www.reddit.com/r/openwrt/comments/b51z6b/redirect_only_netflix_traffic_outside_the_vpn/

I am wondering if something similar can be achieved with
gnuton/asuswrt-merlin?

Thanks

jwh75, 2023-06-20:
Tested Welcome to docs.openwrt.melmac.net! | stangri’s OpenWrt Packages
Documentation.
<https://docs.openwrt.melmac.net/pbr/#how-to-install-dnsmasq-full> on a
RT-AC51U...

seems to work perfectly. Policies based on (remote) domain names seems to
work perfectly. I am wondering if anything similar is possible on
asuswrt-merlin, and in that case if it is planned to be done

Thanks