gnuton/asuswrt-merlin.ng #422: 06/19/2023 - Asus security update

megamorf, 2023-06-20:
Asus has just released a bunch of security updates and I wanted to ask if those are already part of or scheduled to be included in the next release:

https://www.asus.com/content/asus-product-security-advisory/

***

<div id="header44" class="sec-title" style="cursor: pointer; color: rgb(51, 51, 51); font-family: &quot;Segoe UI&quot;, Arial, Helvetica, sans-serif; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><strong style="font-family: TTNormsProMedium, MyriadSemibold;">New firmware with accumulate security updates for GT6/GT-AXE16000/GT-AX11000 PRO/GT-AXE11000/GT-AX6000/GT-AX11000/GS-AX5400/GS-AX3000/XT9/XT8/XT8 V2/RT-AX86U PRO/RT-AX86U/RT-AX86S/RT-AX82U/RT-AX58U/RT-AX3000/TUF-AX6000/TUF-AX5400</strong><span> </span>∇</div><div id="body44" class="sec-desc" style="padding: 5px 0px; display: block; color: rgb(51, 51, 51); font-family: &quot;Segoe UI&quot;, Arial, Helvetica, sans-serif; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><p style="font-weight: 400; margin: 0px;">We strongly encourage you to periodically audit both your equipment and your security procedures, as this will ensure that you will be better protected. As a user of an ASUS router, we advise taking the following actions:</p><ol style="padding: 0px; margin: 0px;"><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Update your router to the latest firmware. We strongly recommend that you do so as soon as new firmware is released. You will find the latest firmware available for download from the ASUS support page at <a href="https://www.asus.com/support/" style="color: rgb(51, 185, 255); text-decoration: none; transition: all 0.2s ease-in-out 0s; background-color: rgba(0, 0, 0, 0);">https://www.asus.com/support/</a>or the appropriate product page at <a href="https://www.asus.com/Networking/" style="color: rgb(51, 185, 255); text-decoration: none; transition: all 0.2s ease-in-out 0s; background-color: rgba(0, 0, 0, 0);">https://www.asus.com/Networking/</a>. ASUS has provided a link to new firmware for selected routers at the end of this notice.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Set up separate passwords for your wireless network and router-administration page. Use passwords with a length of at least eight characters, including a mix of capital letters, numbers and symbols. Do not use the same password for multiple devices or services.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Enable ASUS AiProtection, if your router supports this feature. Instructions on how to do this can be found in your router’s manual, or on the relevant ASUS support page, at <a href="https://www.asus.com/Networking/" style="color: rgb(51, 185, 255); text-decoration: none; transition: all 0.2s ease-in-out 0s; background-color: rgba(0, 0, 0, 0);">https://www.asus.com/Networking/</a>.</li></ol><p style="font-weight: 400; margin: 0px;"> </p><p style="font-weight: 400; margin: 0px;">Please note, if you choose not to install this new firmware version, we strongly recommend disabling services accessible from the WAN side to avoid potential unwanted intrusions. These services include remote access from WAN, port forwarding, DDNS, VPN server, DMZ, port trigger.<br><br>For further help with router setup and an introduction to network security, please visit<br><a href="https://www.asus.com/support/FAQ/1008000" style="color: rgb(51, 185, 255); text-decoration: none; transition: all 0.2s ease-in-out 0s; background-color: rgba(0, 0, 0, 0);">https://www.asus.com/support/FAQ/1008000</a><br><a href="https://www.asus.com/support/FAQ/1039292" style="color: rgb(51, 185, 255); text-decoration: none; transition: all 0.2s ease-in-out 0s; background-color: rgba(0, 0, 0, 0);">https://www.asus.com/support/FAQ/1039292</a></p><p style="font-weight: 400; margin: 0px;"> </p><p style="font-weight: 400; margin: 0px;">The new firmware incorporates the following security fixes.</p><ol start="1" style="padding: 0px; margin: 0px;"><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed CVE-2023-28702, CVE-2023-28703, CVE-2023-31195, CVE-2022-46871, CVE-2022-38105, CVE-2022-35401, CVE-2018-1160, CVE-2022-38393, CVE-2022-26376</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed DoS vulnerabilities in firewall configuration pages.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed DoS vulnerabilities in httpd.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed information disclosure vulnerability.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed null pointer dereference vulnerabilities.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed the cfg server vulnerability.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed the vulnerability in the logmessage function.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed Client DOM Stored XSS</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed HTTP response splitting vulnerability</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed status page HTML vulnerability.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed HTTP response splitting vulnerability.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed Samba related vulerabilities.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed Open redirect vulnerability.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed token authentication security issues.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Fixed security issues on the status page.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Enabled and supported ECDSA certificates for Let's Encrypt.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Enhanced protection for credentials.</li><li style="list-style: unset; line-height: 20px; display: list-item; text-align: -webkit-match-parent; padding: 0px; margin: 0px;">Enhanced protection for OTA firmware updates.</li></ol><p style="font-weight: 400; margin: 0px;"> </p><p style="font-weight: 400; margin: 0px;"> </p>

Model name | Firmware download path
-- | --
GT6 | https://rog.asus.com/networking/rog-rapture-gt6-model/helpdesk_bios/
[... cut ...]
RT-AX58U | https://www.asus.com/networking-iot-servers/wifi-routers/asus-wifi-routers/rt-ax58u/helpdesk_bios/?model2Name=RT-AX58U
RT-AX3000 | https://www.asus.com/us/networking-iot-servers/wifi-routers/asus-wifi-routers/rt-ax3000/helpdesk_bios/?model2Name=RT-AX3000
TUF-AX6000 | https://www.asus.com/networking-iot-servers/wifi-routers/asus-gaming-routers/tuf-gaming-ax6000/helpdesk_bios/?model2Name=TUF-Gaming-AX6000
TUF-AX5400 | https://www.asus.com/networking-iot-servers/wifi-routers/asus-gaming-routers/tuf-gaming-ax5400/helpdesk_bios/?model2Name=TUF-Gaming-AX5400

</div>

Nilssonfpv, 2023-06-21:
This should be prio 1 on the todo list 😅

CVE-2022-26376 Is Baad, ranked 9.8/10 on the risk scale.

gnuton, 2023-06-22:
@Nilssonfpv correct beside that also the issue from 2018 was fixed in an old release that I know.
In any case we are in touch with ASUS and the new GPLs should come soon.

stmuk, 2023-10-25:
I was wondering what the status of this was and whether upgrading to stable 388.2_2_0 or an unstable release is recommended?

gnuton, 2023-10-25:
most of the issues here were already fixed in Merlin. If you wanna get covered about the latest issue, jsut install the beta2 which is stable enough