gnuton/asuswrt-merlin.ng #423: Unable install on Asus AX-82U

bdkacz, 2023-06-22:
**Router Model Affected**
Models: RT-AX82U v1

**Firmware Version Affected**
388.2.2

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
I'm unable to flash the firmware  from factory to this 
[image]
  using webui - I get  "please wait updating" and no progress bar

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Administration
2. Click on Update firmware
3. Klick browse
4. Select file and start flashing

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

bdkacz, 2023-06-22:
I also try install once again stock firmware and after a while the flashing bar is displayed and flashed successfully. Next I try one again upload this firmware and..the same stuck on "Please wait"

gnuton, 2023-06-22:
are you sure you have a v1?

bdkacz, 2023-06-22:
Yes, I checked x2 -  on label H/W ver 1.0

When I downgrade to 3.0.0.4.388.22525 and then try flash Your firmware the progress bar goes to 100% and after re-login there is still stock firmware

bdkacz, 2023-06-22:
Ups...

On top of  label : 

Model rt-ax82u v2

But near S/N I have  H/W Ver: 1.0 LOL

closed 2023-06-22 by gnuton