gnuton/asuswrt-merlin.ng #424: Administration - Firmware Upgrade Bad Link

phostetl, 2023-06-24:
**Router Model Affected**
RT-AXE95Q

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
N/A

**Describe the bug**
The link in note #4 on the web interface goes to https://www.asuswrt-merlin.net/download instead of gnuton's stable builds on github.

**To Reproduce**
Steps to reproduce the behavior:
1. Click the link and get Merlin update site, not Gnuton's site.
4. See the error

**Expected behavior**
The link should take you to gnuton stable releases.

**Screenshots**

gnuton, 2023-06-28:
Thanks for reporting I will check this.

gnuton, 2023-10-04:
This should be fixed in the latest beta

gnuton, 2023-10-04:
duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/467

closed 2023-10-04 by gnuton