gnuton/asuswrt-merlin.ng #431: In China, it is recommended to add STUN servers

ggllioio, 2023-07-11:
Operator China Mobile。

UPNP is located behind operator NAT, log lot 
Jul 11 23:22:50 miniupnpd [1985]: private/reserved address 10.32.134.237 is not suitable for external IP
Jul 11 23:24:52 miniupnpd[1985]: private/reserved address 10.32.134.237 is not suitable for external IP
。 The query appears to be /usr/sbin/getrealip .sh not getting an IP address.

https://www.snbforums.com/threads/miniupnpd-outdated-doesnt-work-when-wan-is-a-private-address.80653/

China is stun.l.google.com:19302 stun.stunprotocol.org difficult by default。

It is recommended to increase stun.qq.com and stun.miwifi.com。

I'm really sorry I don't speak English well

gnuton, 2023-07-11:
Hi!
If I got it right you are suggesting to replace 
stun.l.google.com:19302 stun.stunprotocol.org 
with
 stun.qq.com and stun.miwifi.com
 in
./release/src/router/rom/apps_scripts/getrealip.sh

is this correct?

ggllioio, 2023-07-12:
> 你好！如果我做对了，你建议用 ./release/src/router/rom/apps_scripts/getrealip 中的 stun.qq.com 和 stun.miwifi.com 替换 stun.l.google.com:19302 stun.stunprotocol.org.sh
> 
> 这是对的吗？
> 
> 作为参考，这里是STUN服务器的公共列表 https://gist.github.com/mondain/b0ec1cf5f60ae726202e?permalink_comment_id=3958696

Yes, it is better to add these two domains. They belong to QQ and Xiaomi. Many entries in the China stun list are inaccessible

linnvirgo, 2024-06-01:
> Operator China Mobile。
>
> UPNP is located behind operator NAT, log lot Jul 11 23:22:50 miniupnpd [1985]: private/reserved address 10.32.134.237 is not suitable for external IP Jul 11 23:24:52 [...]

can you tell me the specific code commands and procedures? i'm china mobile broadband too.