gnuton/asuswrt-merlin.ng #433: Asus XT8: Using LAN port for internet to free the 2.5 port

psande, 2023-07-15:
Hi,

I have an Asus XT8 and I would love to be able to use a Lan 1gbps port for the internet. This is so I can free up the 2.5gbps port for the NAS.

Is this possible or a hardware limitation?

I tried to do this using the Dual WAN feature, I get this message: "WAN port should be selected in Dual WAN."

Thanks in advance.

psande, 2023-07-15:
I found this reading: https://github.com/gnuton/asuswrt-merlin.ng/issues/312

But there is no answer so far.

gnuton, 2023-07-18:
duplicate of #312

closed 2023-07-18 by gnuton

psande, 2023-07-25:
@gnuton I understand this is a duplicate, but do you know if it is possible? Thanks

gnuton, 2023-07-26:
Not sure at moment. I have some ideas that have to be tried out.

psande, 2023-08-01:
Thanks for your answer @gnuton. And thanks for your work. :)