gnuton/asuswrt-merlin.ng #435: Model XT8 and RT-AX82U both of them have aproblem with link as AI Mesh Mode.

hunterpl, 2023-07-18:
**Router Model Affected**
Models: (eg: RT-AX82U and XT8)

**Firmware Version Affected**
388.4_0-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
After upgrade one router main first, so which was RT-AX82U was upgraded successfully, everything was fine, but, when I started upgrade the second router, which was linked as a node, it was successfully upgraded but does not started any more, just blinking blue all the time, I just made a several test with both of routers and switch them even from the scratch [ reset to hwarware default ], the result is always the same, they can not be linked with AI Mesh Mode, not possible, but when I configure them separetly as normal router everything works as expected, only not possible link as AI Mesh Mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to asus router normal login
2. Click on AIMesh
3. Add AIMesh node as new router, even if it was blinked I tried reset to default, same result can not be pair, all the time failed.

**Expected behavior**
I made XT8 as primary router and working fine, but when I wanted to make AX-82U as second node in AI Mesh mode, does not work, all the time blinked, broken config or something, but firmware is installed succefully.

**Screenshots**
there is no need a screenshots.

gnuton, 2023-07-19:
Thanks I will look at what's wrong.
The problem this alpha firmware seems to not work for mesh nodes.  Please correct me if I am wrong.
Question: are the master and node connected with a cable or via wireless?

hunterpl, 2023-07-19:
> Thanks I will look at what's wrong. The problem this alpha firmware seems to not work for mesh nodes. Please correct me if I am wrong. Question: are the master and node connected with a cable or via wireless?

Yes Mesh node does not work only tested for wireless, did not tested with lan, but I think will be the same results.

gnuton, 2023-07-19:
actually it's not the same, binding two wireless devices is way more complex than connecting two devices connected through ethernet. Usually there is not packet encryption over ethernet nor scanning and pairing.. over ethernet it's mostly plug and play.

hunterpl, 2023-07-19:
Yes this is correct information, but this is not possible if first attempt is configuration via phone and use app from asus router via WIFI, so if WIFI does not work I can not switch to Lan :C

gnuton, 2023-07-19:
Asus app is not supported. It may work, but it's not supported.
I never tried to pair via Wifi and then switch to ethernet, but the other way around works.

JonnyVR1, 2023-07-20:
I've tried pairing with both wire and wireless, neither work on 388.4, even for nodes that existed before the update, they show as offline

JonnyVR1, 2023-09-24:
Works as of the latest beta, @gnuton this can be closed

hunterpl, 2023-09-24:
Yes working for the latest firmware, great job.

gnuton, 2023-09-24:
👍 awesome I will close this.

closed 2023-09-24 by gnuton