gnuton/asuswrt-merlin.ng #436: 120429

micron888, 2023-07-22:
Models:TUF-AX5400 v1

**Firmware Version Affected**
[Unstable: 388.4_0-gnuton0_alpha1]

**Is this bug present in upstream Merlin releases too?**
388.2_2 is ok

**Describe the bug**
network map showing 0 devices

Roki100, 2023-07-22:
same on RT-AX82U, also running 388.4_0-gnuton0_alpha1]

micron888, 2023-09-12:
Gnuton,
Do you mind sharing the image :)

gnuton, 2023-09-13:
https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1592/workflows/ddd141b0-01ed-496c-8b45-b7a6b4835fce
click the model then "artifact"

Roki100, 2023-09-13:
can confirm the issue is gone on latest build linked above

micron888, 2023-09-13:
latest build resolve network client map issue.
[image]

micron888, 2023-09-14:
Web login outside wifi not working.
Log show connection closed unexpectedly

gnuton, 2023-09-15:
Could you please open a new ticket for different issues? Otherwise they are
hard to be tracked

gnuton, 2023-09-23:
closing this since fixed

closed 2023-09-23 by gnuton