gnuton/asuswrt-merlin.ng #438: 388.4_0-gnuton0_alpha1 main router client counts, lists, and statuses showing 0/nothing/empty

shiang-luong, 2023-07-27:
**Router Model Affected**
Models: xt8

**Firmware Version Affected**
388.4_0-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
The client list isn't showing up or being populated.  Dirty upgrade from 388.2_2_0-gnuton1

**To Reproduce**
Steps to reproduce the behavior:
1. Log into Asus router
2. On main home page you will see 0 for clients
3. When you click on the client 'view list', there is no clients listed
4. When you click on number 0, no clients are listed on the right hand side under 'Client Status'

**Expected behavior**
User should be able to see list of clients in the views mentioned above

**Screenshots**
[image]

tclam662, 2023-07-30:
Got the same problem with my xt8 on 388.4_0-gnuton0_alpha1, have to revert it back to 388.2.2

shiang-luong, 2023-07-31:
> Got the same problem with my xt8 on 388.4_0-gnuton0_alpha1, have to revert it back to 388.2.2

I'm glad it isn't just me.

disconnectome, 2023-08-03:
ax5400 same, ok

micron888, 2023-09-23:
latest beta is ok
Release: 3004.388.4_0-gnuton0_beta1

shiang-luong, 2023-09-29:
Agreed, this can be closed.  It is fixed in the beta

closed 2023-09-30 by gnuton