gnuton/asuswrt-merlin.ng #439: [Question] ET8 and 388.4_0-gnuton0_alpha1

lorenc-tomasz, 2023-07-29:
Hi,
I don't see FW for ET8 within 388.4_0-gnuton0_alpha1 release. Only log file? Do you plan to release 388.4_0 for ET8?

gnuton, 2023-07-30:
I noticed that. The only reason why it a missing is because I got a faulty
GPL from Asus which was missing some files.

DiMENZi0N, 2023-08-04:
ASUS finally went ahead and released some new official firmware for the ET8 8-03-23.

https://www.asus.com/us/supportonly/zenwifi%20et8/helpdesk_bios/

gnuton, 2023-08-10:
I fixed the build for ET8 and it's all fine.
I have the firmware again.It builds locally without problem.

Usually I build the firmware in the cloud, because it's more transparent and reliable.  The problem now is that it takes more than 1 hour for the firmware to build in the cloud and circleci https://twitter.com/gnuton/status/1686319259301838848 which limits the builds to 1-hour makes those builds fail all the time.

If you can please make some pressure to circle ci on twitter to relax their policy

CPngN, 2023-08-10:
Oh no! My damn ET8s, always a special exception! LOL

FlavioDF, 2023-08-15:
Do we have any changes to have the firmware of the ET8? Thanks in advanced

gnuton, 2023-08-16:
ATM all ET8 changes are on github. I just merged Beta3.

gnuton, 2023-09-25:
ET8 images are available again.
Closing this

closed 2023-09-25 by gnuton