gnuton/asuswrt-merlin.ng #440: AiMesh Adding Node failed

tclam662, 2023-07-30:
**Router Model Affected**
RT-AX95Q

**Firmware Version Affected**
388.4_0-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
AiMesh Adding Node failed:
388.4_0-gnuton0_alpha1 Router (RT-AX95Q) + 388.4_0-gnuton0_alpha1 Node (RT-AX95Q) Failed, Node could be detected but adding node failed

388.2_2-gnuton1 Router (RT-AX95Q)  + 388.4_0-gnuton0_alpha1 Node (RT-AX95Q) Failed, Node could be detected but adding node failed

**To Reproduce**
Steps to reproduce the behavior:
1. Have RT-AX95Q Node and Router upgraded to 388.4_0-gnuton0_alpha1, reset both router and nodes after upgrade.
2. Config 1 of the RT-AX95Q as AiMesh Router and another one as Aimesh node
3. On the AimeshRouter, add AiMesh Node. The node could be detected after the scan, click Add
4. Wait til The progress bar finishes and and the message "Unable to add Aimesh node xxxxx" appears.

**Expected behavior**
1. Config 1 of the RT-AX95Q as AiMesh Router and another one as Aimesh node
2. On the AimeshRouter, add AiMesh Node. The node could be detected after the scan, click Add
3. Wait til The progress bar finishes and AiMesh node successfully added.

gnuton, 2023-07-30:
Hi,
This is a known issue. Can you connect the two routers via Ethernet cable
and try to pair the two ai mesh node again?

Dodgydrains, 2023-08-29:
> I can confirm this issue is also on Internet Cable. So Alpha 1 is not usable with AIMesh nodes.

Same for me.  I could not add a node via ethernet cable either.  Had to use ASUS Firmware Restoration to revert to official firmware to use AiMesh

gnuton, 2023-09-25:
The beta should have fixed thi. Can you confirm?

asterictnl-lvdw, 2023-09-25:
Can confirm after updating to the latest beta the issue has been resolved. My node is back online and I can use the build again.

Thank you~

gnuton, 2023-09-25:
awesome thanks @asterictnl-lvdw closing this issue.

closed 2023-09-25 by gnuton