gnuton/asuswrt-merlin.ng #443: Record removed when clicking edit icon in MAC/IP DHCP binding form

anst84, 2023-08-02:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Pressing the "edit icon" to an existing row in the "Manually Assigned IP around the DHCP list (Max Limit : 64)" table makes the row disappear. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **Advanced Settings**
2. Click on **LAN**
3. Click on tab **DHCP Server**
4. Scroll down to table with header named **Manually Assigned IP around the DHCP list (Max Limit : 64)**
5. Click on edit icon on existing record/row

**Expected behavior**
Edit the record

**Screenshots**
Screenshot N/A but I can upload video somewhere if helps. Thanks in advance!

anst84, 2023-08-02:
I just realized that when clicking this it actually fills in the MAC in the top form and makes it ready to edit and reapply with save. Maybe this is the intended way of editing. If so I would like converting it to a feature request to allow for "inline editing" instead :-)

gnuton, 2023-09-25:
closing this sincei it's not a bug

closed 2023-09-25 by gnuton