gnuton/asuswrt-merlin.ng #448: Asus TUF 5400

ForumDyskusyjne, 2023-08-23:
Factory firmware has problems:
1.)
With memory management.
**After 27 days without "restart" router has problem with services stability.**
Router decreases free RAM and increases processors use.
2.)
Not works correctly adaptive QOS.

**Those problems ther are in the merlin?**

gnuton, 2023-08-29:
which version?

ForumDyskusyjne, 2023-08-30:
ASUS TUF-AX5400 3.0.0.4.388.23285 
https://www.snbforums.com/threads/asus-tuf-ax5400-firmware-version-3-0-0-4-388-23285.85053/

I have open report to support. All described bugs are described and contain full logs from router.
At the moment I have to wait 21 days, because I promised full logs for support. I send full log what 7 days.
I am interested Merlin's, has the same problem?

gnuton, 2023-09-25:
This is stock firmware and not a gnuton's one.
Please reach Asus support.

closed 2023-09-25 by gnuton