gnuton/asuswrt-merlin.ng #449: Where is a Beta Version

neil1962, 2023-08-28:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
Latest Alpha

**Is this bug present in upstream Merlin releases too?**
Cant test, but I dont see repoted issued in Merlins firmware using the the same ASUS GPL accross routers

**Describe the bug**
Many, my 2 cents worth, Network listing, does not work, no clients are being listed, always blank, can though see clients via wireless log.  Cannot bind a client to a node. Connecting nodes is hit and miss even using an Ethernet connection. On upgrading from gnuton current stable release to latest alpha, corruption occurred in the client names in the network listing before all client names just disappeared, and became blank on reboot of the router. Not stable, all my nodes are using current ASUS stock firmware, constant drops with clients, reverted to latest current gnutom, all stable again. Have tried multiple WPS resets, clearing browers cache, etc, does not fix the problems.I have to say this is the worst Alpha firmware I have seen from gnuton. I have been keeping my eye with different issues being reported with the latest Alpha firmware release and I am also not seeing any response from gnuton for a while now for these reported for issues.

dacootmeister, 2023-08-28:
following this thread.... have not upgraded from 388.2_2-gnuton1 bcause of reported issues

gnuton, 2023-08-29:
Thanks for testing @neil1962.
I will try to address al these issue. The beta will come soon.

gnuton, 2023-08-29:
@neil1962 This a beta image for RT-AX82U V1. 
https://output.circle-artifacts.com/output/job/af759e87-352d-491a-8766-9d6aa96ea7c7/artifacts/0/workspace/release/RT-AX82U_3004_388.4_beta3_puresqubi.w 

I am not connected through it and it works fine after a dirty flash.
I have not tested AI Mesh although.

dacootmeister, 2023-08-31:
> Thanks for testing @neil1962. I will try to address al these issues. The beta will come soon.

Thanks, will keep a close eye on it for the ZenWiFi_XT8 (base model: RT-AX95Q)

shiang-luong, 2023-08-31:
I am also patiently waiting for the new beta ZenWiFi_XT8 (base model: RT-AX95Q)

jef00, 2023-09-07:
[image]
the name scheme is some different because i compiled myself is that correct @gnuton ? Its runs smooth on the RT-AX95Q!!

shiang-luong, 2023-09-07:
> [image]
> the name scheme is some different because i compiled myself is that correct @gnuton ? Its runs smooth on the RT-AX95Q!!

If it works, I'm going to need to try that!!  Thanks

jef00, 2023-09-07:
no problem @shiang-luong

jef00, 2023-09-08:
the name scheme is changed that was stupid of me it is in the changelog. @gnuton question if i installed the compiled version is it then possible to check for firmware updates?

gnuton, 2023-09-08:
The new schema version change comes from upstream. The ability to  look for updates is made of client and server parts. I need to test if it properly works. So building from source code is not an issue for that.

jef00, 2023-09-08:
@gnuton thankyou for you response!! aah yes i see!. take your take for testing that out,  I am asking because i have built it and installed but if i want to check for updates i get this error:

[image]

maybe i did something wrong when i compiled, i did 
#cd release/src-rt-5.02axhnd.675x
#make rt-ax95q

can you tell me if i miss something sorry to bother you

gnuton, 2023-09-08:
I just need to fix something there. That was low prio, I noticed that last night, again, I have to fix it. The entire code for the version checking has changed due to new versioning schema. As for the timeline for the fix, it will be one of the last things I will fix before releasing the stable

jef00, 2023-09-08:
@gnuton i can completely understand! thankyou for explaining and take your time!:)

gnuton, 2023-09-25:
the beta is out. 
Closing this

closed 2023-09-25 by gnuton