gnuton/asuswrt-merlin.ng #450: Is there even a v2 of TUF-AX5400?

gameinn1, 2023-08-29:
The site states that v1 of TUF-AX5400 is supported but is there even a v2 of this router? I would recommend only adding stuff like "v1" if there are known multiple versions to limit confusion.

gnuton, 2023-08-29:
Which site?
The main page states the supported routers with their versions
[image]

gameinn1, 2023-08-30:
Yes and that's the main point.

Do you know of a TUF-AX5400 v2 in existsence via box photo or bottom of the router stating v2? It just confuses people putting "TUF-AX5400 v1" instead of "TUF-AX5400" making people think different versions exist when they don't.

If anything you should put vA instead as this is what the TUF line has on the bottom: https://i.imgur.com/Ns7KAPH.png and https://i.imgur.com/AxDF4ds.jpeg

and on the box: https://i.imgur.com/1xfZpBo.png

gnuton, 2023-09-03:
I am not sure if it's reliable to see the version of the router from the box.
I have wrote in the page the way you should get this info. Please see the highlighted text.

[image]

iamromulan, 2023-09-14:
> Yes and that's the main point.
>
> Do you know of a TUF-AX5400 v2 in existsence via box photo or bottom of the router stating v2? It just confuses people putting "TUF-AX5400 v1" instead of "TUF-AX5400" [...]

Mines a V2
[image]

Karabacak021, 2024-01-19:
The main issue is, that if I follow the instructions, I still don't know it is V1 or not.
According to [this forum](https://www.snbforums.com/threads/correctly-merlin-for-tuf-gaming-ax5400.86435/) H/W Ver. A + FCC ID: MSQ-RTAX8D00 is the V1. 

@iamromulan , May can you share you H/W version? If it is not A, then ideal will be:
TUF-AX5400 vA or TUF-AX5400 A
TUF-AX5400 vB or TUF-AX5400 B (If romulans will be B)

[image]

iamromulan, 2024-01-23:
> The main issue is, that if I follow the instructions, I still don't know it is V1 or not.
> According to [this forum](https://www.snbforums.com/threads/correctly-merlin-for-tuf-gaming-ax5400.86435/) [...]

Sure, next time I am in the same room with it I'll take a look. I was wanting this firmware to try and get Zerotier or tailscale running on it but since I was able to install tailscale directly to the cellular modem that feeds it I moved on 🤣

znre, 2024-07-12:
[image]

If it only says "TUF-AX5400", is it v1?

inigortega, 2025-02-28:
> [image]
> 
> If it only says "TUF-AX5400", is it v1?

How can I execute this command? Via ssh?