gnuton/asuswrt-merlin.ng #452: Slave nodes unavailable for firmware upgrade from Console UI.

DiMENZi0N, 2023-08-30:
**Router Model Affected**
Models: Asus RT-AXE95Q (Zenwifi ET8)

**Firmware Version Affected**
388.4_beta3

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Firmware upgrade for slave nodes in Mesh network is not present.

**To Reproduce**
Steps to reproduce the behavior:
Login to Master Console.
Goto Firmware upgrade page.
ET8 AiMesh Slave is not present to select for upgrade.

**Expected behavior**
I expect to see my attached slave nodes to be available to flash firmware from the Master UI.

**Screenshots**
[image]

DiMENZi0N, 2023-09-27:
Heads up this issue still remains in new firmware

[image]

gnuton, 2023-09-30:
fixed 72db9aa48ec3e873d0158239f30b6a629186f21f

closed 2024-02-11 by gnuton