gnuton/asuswrt-merlin.ng #453: Nodes offline and in configure mode.

goalieman86, 2023-09-07:
**Router Model Affected**
Models: ZenWifi XT8 (RT-AX95Q

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
After update to my router/nodes everything seemed stable and working great. After a period of about 2mo on the firmware my nodes became unstable and eventually just sat there with a flashing blue LED. After a reboot it would come back for a short time with the white LED, but then kick back to a flashing blue LED. I tried reuploading the firmware and resetting the devices and reconnecting the nodes to the network to no avail. This morning I rolled back the firmware to 388.1_0-gnuton1 and things are back to normal.  Wired connection was unaffected by this issue, I just didn't have the node as an AP.

**To Reproduce**
Steps to reproduce the behavior:
No steps needed. The nodes crashed and would not recover once in "configure" mode. 

**Expected behavior**
Typically a reset of the device would clear any issues and the system would work. This did not happen after several attempts and even a reflash of the latest FW. Only suitable result was rolling back the firmware .

dacootmeister, 2023-09-07:
Familiair situation which occurred last week at my place as well with the same firmware version.
I do a reboot cycle every night of my mesh system, to keep it fresh and responsive.

Luckily I could restore my config from a backup file I made recently and was back in business within the hour.
It's strange to encounter configure mode while waking up and none of your devices are reachable, but you do have Wifi signal.

gnuton, 2023-09-08:
Theoretically if  388.2_2_0-gnuton1 worked for 2 months it should be fine.
If it stopped to work after 2 months, something happened. No idea what, you may know bette then me in this case.
My suggestion is to stock firmware on the nodes and Merlin on the main unless there is a specific reasaon why you wanna install merlin everywhere.
The entire AIMEsh thing is closed source. So there is no so much I can do about it.

dacootmeister, 2023-09-09:
> Theoretically if 388.2_2_0-gnuton1 worked for 2 months it should be fine. If it stopped to work after 2 months, something happened. No idea what, you may know bette then me in this case. My suggestion is to stock firmware on the nodes and Merlin on the main unless there is a specific reasaon why you wanna install merlin everywhere. The entire AIMEsh thing is closed source. So there is no so much I can do about it. But I would be glad to help if there is anything I can do.

In my case the main node was dropping back to config mode, no idea what the mesh nodes were doing at that point. And I am rebooting on a daily basis.

Ah well, lets hope for the new beta soon ey, @gnuton  :-) Tnx