gnuton/asuswrt-merlin.ng #454: RT-AX82U will not upgrade

Enigma1282, 2023-09-08:
**Router Model Affected**
ModelsRT-AX82U
The web UI, after i attempt to upgrade to the merlin firmware, the UI will say "Please wait... Applying settings" and just stick there. nothing else will happen.

gnuton, 2023-09-08:
Can you show me the output of 
nvram get productid

Enigma1282, 2023-09-08:
RT-AX82U_388.2_2_0-gnuton1_puresqubi.w is the first one i attempted. I also attempted to use RT-AX82U_388.4_0-gnuton0_alpha1_puresqubi.w. Also I may sound dumb but what is " the output of
nvram get productid"

elliotberry, 2023-09-08:
Not dumb at all. i just had to figure it out myself. Enable SSH on the router first, then SSH in and run that command. People say "nvram get productid" and they don't explain any context whatsoever

Enigma1282, 2023-09-09:
"rt-ax82u_v2" is what the command brings back.

hammerjoe, 2023-09-10:
> "rt-ax82u_v2" is what the command brings back.

It looks like you have V2 of the router and the image available is for V1. I think there is a V2 image somewhere.

gnuton, 2023-09-11:
V2 is not supported yet. 
It will be supported soon btw. I already have working images, they are just not public yet.

@elliotberry good point. I will improve the documentation. thanks for pointing out!

closed 2023-09-11 by gnuton