gnuton/asuswrt-merlin.ng #457: TUF-5400 VPN tabs empty

darkdex52, 2023-09-14:
**Router Model Affected**
Models: [TUF-AX5400 v1](https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/TUF-Gaming-AX5400/)

**Firmware Version Affected**
[388.4_0-gnuton0_alpha1](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/388.4_0-gnuton0_alpha1)

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
After updating from 3.88.2 to 3.88.4, VPN Server and VPN Client tabs are completely empty and non-interactable.

**Screenshots**
https://i.imgur.com/Pm7SZoD.png
https://i.imgur.com/nJ4d5dU.png

gnuton, 2023-09-25:
is this fixed in the latest beta?

master88wi, 2023-09-25:
> is this fixed in the latest beta?

no

gnuton, 2023-09-25:
Problem in VPN Server page
[image]

gnuton, 2023-09-25:
Problem in VPN CLient page
[image]

gnuton, 2023-09-27:
Tested. There is still a bug here
[image]

gnuton, 2023-09-27:
Another issue in http://router.asus.com/Advanced_WireguardClient_Content.asp
[image]

darkdex52, 2023-10-05:
Fixed in [3004.388.4_0-gnuton0_beta2](https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/3004.388.4_0-gnuton0_beta2)

closed 2023-10-05 by darkdex52