gnuton/asuswrt-merlin.ng #458: Can't install Merlin to ZenWiFi XT8 v2

sanjeroHUN, 2023-09-14:
**Router Model Affected**
Models: ZenWiFi XT8 v2

**Firmware Version Affected**
3.0.0.4.388_23285

**Describe the bug**
After selecting the RT-AX95Q_388* Merlin firmware for upload to my Zenwifi XT8 (V2) router the "Please wait, applying settings" title appears, but nothing else happens.
Am I right, that the Merlin version is currently not compatible with the V2 version of the Zenwifi XT8 router? Please try to make it compatible. :)

Thank you!

Dodgydrains, 2023-09-14:
I don't think the V2 is compatible with Merlin, only V1

gnuton, 2023-09-15:
Correct only V1 is supported for now

sanjeroHUN, 2023-09-17:
Thanks! I will send it back due to the high latency peaks... For gaming it is unusable...

gnuton, 2023-09-25:
closing this since it' a new model request.

closed 2023-09-25 by gnuton