gnuton/asuswrt-merlin.ng #460: App Connection Issue

micron888, 2023-09-16:
Models:TUF-AX5400 v1

Firmware Version Affected
DEV_3004.388.4+RT-AX82U_V2_TEST_PARALLEL2

Is this bug present in upstream Merlin releases too?
388.2_2 is ok

Describe the bug
Asus Router App cannot connect from WAN
LAN ok

gnuton, 2023-09-16:
If I got it right you are saying the web interface is not visible from a
public IP. Did I get it right?

micron888, 2023-09-16:
I am able to access the login page from wan but run into login issues on both web or Asus app.

micron888, 2023-09-16:
[image]

gnuton, 2023-09-17:
what about accessing the router from WAN with a browser? does it work?

micron888, 2023-09-17:
Nope, it doesn't work
login page keep showing up after I signin

gnuton, 2023-09-17:
Did you see anything in the logs? Have you tried to reset to factory? Does it help? This feels to be something related to something broken due to settings which are not compatible with the new binaries

micron888, 2023-09-18:
I will do a reset this weekend, and report finding

micron888, 2023-09-20:
Ir seem to work magically today without doing anything.. Closing

closed 2023-09-20 by micron888