gnuton/asuswrt-merlin.ng #461: Enable WAN setting not correct after disconnection via Internet Status top menu.

Yakumounr, 2023-09-18:
**Router Model Affected**
Model: DSL-AX82U

**Firmware Version Affected**
388.4_0-gnuton0_alpha1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Used the Internet Status icon top right of the top bar, single click, select 'Disconnect the internet connection' option to disconnect (successfully).
Went to WAN settings, opened the connection, Enable WAN was still showing with 'Yes' selected. Connection re-established only after moving Enable WAN to 'No' and applying.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Internet Status' and click once
2. Click on 'Disconnect the internet connection'
3. Go to WAN settings, open connection, check status of 'Enable WAN' radio buttons.
4. Toggle Enable WAN to restart the connection.

**Expected behaviour**
Disconnecting should have caused the Enable WAN setting to switch to No, leaving moving it to Yes to re-establish the connection.

gnuton, 2023-09-19:
disconnecting the internet from the WAN from the button in the bar makes the WAN connection going in "Hot standby"
I checked this in the latest firmware I do have and I confirm it works in the way you are describing.
To bring the interface on again i had to switch if off completely (so my understanding is from hot standby -> OFF) and then moving back to YES the wan conection goes to ON again.

At moment I am not sure what's if this is a bug or a feature, What's the behaviour in the stock firmware?

Yakumounr, 2023-09-19:
It's not something that I did very often over the last few years with Asus routers with these settings,  but my recollection is that disconnecting via the status icon as described always resulted in "Enable WAN" then having No selected, and turning that to Yes restarted the connection, which is the logical way for that  options to react.

gnuton, 2023-09-20:
It may be a bug, but it has low priority.
I will take at look at it once I do have some time.