gnuton/asuswrt-merlin.ng #465: Gestione frame protetti (WiFi)

BoostOver, 2023-09-27:
**Router Model Affected**
DSL-AX82U/RT-AX82U

**Firmware Version Affected**
3004.3004.388.4-gnuton0_bet

**Is this bug present in upstream Merlin releases too?**
Non ne sono a conoscenza

**Describe the bug**
L’opzione “gestione frame protetti”(che risulta fondamentale per quanto concerne WPA3) non è presente.
L’opzione permetteva la scelta di tre diverse possibilità:
Possibile,richiesta,disattivata.

**To Reproduce*

http://router.asus.com/Advanced_Wireless_Content.asp

**Expected behavior**
Il ripristino di tale opzione.

gnuton, 2023-09-28:
The text place holder is WLANConfig11b_x_mfp
The code is in Advanced_Wireless_Content.asp
`<tr style="display:none">
                                        <th><#WLANConfig11b_x_mfp#></th>
                                        <td>
                                                <select name="wl_mfp" class="input_option" onchange="handleMFP();">
                                                                <option value="0" <% nvram_match("wl_mfp", "0", "selected"); %>><#WLANConfig11b_WirelessCtrl_buttonname#></option>
                                                                <option value="1" <% nvram_match("wl_mfp", "1", "selected"); %>><#WLANConfig11b_x_mfp_opt1#></option>
                                                                <option value="2" <% nvram_match("wl_mfp", "2", "selected"); %>><#WLANConfig11b_x_mfp_opt2#></option>
                                                  </select>
                                                  <span id="mbo_notice_wpa3" style="display:none">*If the Authentication Method is WPA3-Personal, the Protected Management Frames will be Required.</span>
                                                  <span id="mbo_notice_combo" style="display:none">*If the Authentication Method is WPA2/WPA3-Personal, the Protected Management Frames will be Capable.</span>
                                                  <span id="mbo_notice_combo_legacy" style="display:none">*If the Authentication Method is WPA/WPA2-Personal, the Protected Management Frames will be Capable.</span>
                                                  <span id="mbo_notice" style="display:none"><#WLANConfig11b_AgileMultiband_note#></span>
                                        </td>
                                </tr>
`

gnuton, 2023-09-28:
It looks like ASUS has hidden this on porpouse since this value is automatically set according to the authentication method.

If you wanna check how it change open the chrome/firefox console and type
document.form.wl_mfp.value 

I tested just a few values on my TUF and  this is what I noticed.

0 
1 <-- is set if WPA/WPA2 is selected
2 <-- is set when WPA3 is selected

If you like you can set this value from shell by setting wl_mfp

closed 2023-09-28 by gnuton