gnuton/asuswrt-merlin.ng #466: AddOn FlexQos (Adaptive QoS) non mostra il grafico dell’upload

BoostOver, 2023-09-27:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
Versione corrente : 3004.3004.388.4-gnuton0_beta1 e precedenti

**Is this bug present in upstream Merlin releases too?**
In alcuni casi si

**Describe the bug**
Il grafico in tempo reale dell’upload non è rilevato

**To Reproduce**
https://www.snbforums.com/threads/flexqos-1-3-2-flexible-qos-enhancement-script-for-adaptive-qos.75620/post-835884

**Expected behavior**
https://www.snbforums.com/threads/flexqos-1-3-2-flexible-qos-enhancement-script-for-adaptive-qos.75620/post-836658

P.s.
La nuova chat su Discord non mostra canali attivi.
Grazie

BoostOver, 2023-12-17:
Hai novità in merito a questo problema?
Grazie e ciai

gnuton, 2023-12-18:
Ciao,
sinceramente va visto bene. Il codice linkato con l hardcoding di eth0/4 nn sta da un bel po, lo avevo tolto tempo fa e sostituito con questo:
wan_ifname viene letto da nvram e se vuoto allora fa fallback a eth0.

`int ej_tcclass_dump_array(int eid, webs_t wp, int argc, char_t **argv) {
        FILE *fp;
        int ret = 0;
        char tmp[64];
        int qos_type;
        char wan_ifname[16];

        strlcpy(wan_ifname, nvram_safe_get("wan_ifname"), sizeof(wan_ifname));
        if (*wan_ifname == '\0')
                strlcpy(wan_ifname, "eth0", sizeof(wan_ifname));

        qos_type = nvram_get_int("qos_type");

        if (nvram_get_int("qos_enable") == 0) {
                ret += websWrite(wp, "var tcdata_lan_array = [[]];\nvar tcdata_wan_array = [[]];\n");
                return ret;
        }

`

BoostOver, 2024-03-16:
Ciao, hai avuto tempo di capire dove sta il problema?
Perché anche senza lo script il flusso in upload non viene mostrato.
Grazie