gnuton/asuswrt-merlin.ng #467: Link per il download del firmware (Gnuton) non corretto

BoostOver, 2023-09-27:
**Router Model Affected**
Tutti

**Firmware Version Affected**
Tutti

**Is this bug present in upstream Merlin releases too?**
Non fa testo

**Describe the bug**
http://router.asus.com/Advanced_FirmwareUpgrade_Content.asp
Nell’ultima nota viene riportato il link (a mio avviso errato)….. https://www.asuswrt-merlin.net/download/

**Expected behavior**
Il link corretto dovrebbe essere:
https://github.com/gnuton/asuswrt-merlin.ng

gnuton, 2023-09-27:
Su TUF UI, il check nn va devo controllarlo.

[image]

BoostOver, 2023-09-27:
Si anche da me non funziona, ma non era questo quello che intendevo.
Sono le note al centro della pagina….mi sembra nota 4z

mrbardman, 2023-09-28:
English? Please...

BoostOver, 2023-09-28:
> English? Please...

How i can translate your English you can translate my language.
Don’t stop your brain.

mrbardman, 2023-09-28:
Never mind.

gnuton, 2023-09-28:
Fix ff803ceeabb4d74f8584ba7467da7b6e5ff90d60

BoostOver, 2023-09-28:
[image]
A mio parere la nota 4 non è corretta.
Se uno clicca si ritrova sulla pagina Merlin.

gnuton, 2023-09-28:
[image]
fixed

closed 2023-09-28 by gnuton