gnuton/asuswrt-merlin.ng #468: AiProtection/Allerta preferenze/Google da rimuovere

BoostOver, 2023-09-27:
http://router.asus.com/AiProtection_HomeProtection.asp
In fondo alla pagina a destra è presente un opzione che permette di configurare una mail di contatto in caso si presenti un allert.
Saranno almeno due anni che Google per questioni di sicurezza non permette l’accesso alla propria casella mail tramite il router.
Ci sono spiegazioni ufficiali da parte di Google al riguardo.
A mio avviso sarebbe da eliminare la possibilità di evidenziare Google in tal opzione in modo da evitare di spaccarsi la testa per capire del perché non funzioni.
Mi fa strano che Asus non ci abbia già pensato.
Ciao.

gnuton, 2023-09-27:
mi manderesti il link di questa spiegazione cosi la mando direttamente ad ASUS?

BoostOver, 2023-09-27:
Lo sanno già😃
Leggi:
https://www.asus.com/support/FAQ/1010064/

gnuton, 2023-09-30:
fix 97478e480c3a5b02b5d6dca97e325209e17d4e6f

closed 2023-09-30 by gnuton