gnuton/asuswrt-merlin.ng #469: DSL-AC68U designare una porta lan come wan

BoostOver, 2023-09-27:
Vista “la problematica” della configurazione di una rete AiMesh con firmware Merlin/Gnuton, sarebbe d’aiuto avere una porta lan configurata come wan.
Sarebbe un valore aggiunto alla tua release e non di poco.
Grazie e buon lavoro/hobby :-)

gnuton, 2023-09-27:
concordo in pieno.
Ho gia chiesto in giro come fare, ma al momento nessuno mi ha aiutato. 
Quindi devo fare un po di prove.. ho una vaga idea di dove mettere le mani.. vediamo se funzia..
Ci daro un occhiatta dopo aver rilasicato la stabile dell 388.4

gnuton, 2023-12-03:
@BoostOver Non ancora,  questo modello e' "end of life" ormai da un po, e non so ancora se ci sara' un "prossimo" firrmware, e se ci sara, non so per quanto.