gnuton/asuswrt-merlin.ng #470: Registro wireless non mostra il nome corretto di alcuni apparati connessi

BoostOver, 2023-09-27:
**Router Model Affected**
Tutti

**Firmware Version Affected**
Tutti

**Is this bug present in upstream Merlin releases too?**
Si

**Describe the bug**
Alcuni apparati non vengono mostrati con il loro nome rinominato (da me) come da preferenza individuali ma bensì con il nome reale dell’apparato.

**To Reproduce**
http://router.asus.com/Main_WStatus_Content.asp

gnuton, 2023-09-28:
gentilmente aggiungi l ultima versione del firmware in cui hai trovato il bug,
Inoltre cerca di aggiungere tutti gli step per riprodurre il problema.

gnuton, 2023-09-28:
il MAC address di questi dispositivi e' costante o cambia sempre?

BoostOver, 2023-09-28:
3004.388.4_0-gnuton0_beta1
http://router.asus.com/device-map/clients.asp
Clicco l’icona di default di un qualsiasi dispositivo presente nella mia rete , lo rinomino/cambio l’icona a lui dedicato.
(Non ci sono apparati che cambiano il Mac address come fa la Apple)
Grazie