gnuton/asuswrt-merlin.ng #474: RT-AX82U_V2 not correctly building on circleci

gnuton, 2023-10-04:
As for now this model is build on my machine due some issues with the setup..
This isssue is abotu fixing the ci scripts in order to build working images

gnuton, 2023-10-27:
Locally I get this error related to execstack which can be fixed by using the one in the GPL, but it needs libelf too.

`make -C hostTools/libcreduction install
make[3]: Entering directory '/build/release/src-rt-5.04axhnd.675x/hostTools/libcreduction'
Makefile:94: "32-bit: arm  /opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32"
find: File system loop detected; ‘/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32/usr’ is part of the same file system loop as ‘/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32’.
find: File system loop detected; ‘/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32/usr’ is part of the same file system loop as ‘/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32’.
Makefile:220: All Required 32-bit libraries: libFLAC.so.8 libamas-utils.so libarcher.so libasc.so libasuslog.so libatalk.so.6 libavahi-common.so.3 libavahi-core.so.7 libavcodec.so.58 libavformat.so.58 libavutil.so.56 libbcm.so libbcm_boardctl.so libbcm_crc.so libbcm_flashutil.so libbcm_util.so libbcmmcast.so libbcmtm.so libblkid.so.1 libblobmsg_json.so libblogctl.so libbridgeutil.so libbwdpi.so libbwdpi_sql.so libc.so.6 libcap-ng.so.0 libceshared.so libcfgmnt.so libcharon.so.0 libcodb.so libcom_err.so.2 libconn_diag.so libcrypt.so.1 libcrypto.so.1.1 libcurl.so.4 libdaemon.so.0 libdb-4.8.so libdisk.so libdl.so.2 libe2p.so.2 libebt_arp.so libebt_arpreply.so libebt_ftos.so libebt_ip.so libebt_limit.so libebt_log.so libebt_mark.so libebt_pkttype.so libebt_redirect.so libebt_skiplog.so libebt_standard.so libebt_stp.so libebt_time.so libebt_vlan.so libebt_wmm_mark.so libebtable_broute.so libebtable_filter.so libebtable_nat.so libebtc.so libethctl.so libethswctl.so libev.so.4 libevent-2.0.so.5 libexif.so.12 libexpat.so.1 libext2fs.so.2 libfcctl.so libgcc_s.so.1 libgcrypt.so.11 libgdbm.so.3 libgen_util.so libgpg-error.so.0 libiconv.so.2 libid3tag.so.0 libip4tc.so.0 libip6tc.so.0 libiqctl.so libjansson.so.4 libjpeg.so libjson-c.so.2 libletsencrypt.so liblightsql.so liblldpctl.so.4 liblz4.so.1 liblzo2.so.2 libm.so.6 libmnl.so.0 libmssl.so libncurses.so.6 libneon.so.27 libnetfilter_conntrack.so.3 libnetsnmp.so.35 libnetsnmpagent.so.35 libnetsnmpmibs.so.35 libnfnetlink.so.0 libnl-3.so.200 libnl-genl-3.so.200 libnl-nf-3.so.200 libnl-route-3.so.200 libnsl.so.1 libnt.so libnvram.so libogg.so.0 libovpn.so libpam.so.2 libpasswd.so libpcre.so.1 libpng16.so.16 libptcsrv.so libpthread.so.0 libpwrctl.so libqrencode.so.4 libresolv.so.2 librt.so.1 libshared.so libshn_pctrl.so libshn_utils.so libsmartsync_api.so libsmbclient.so.0 libsqlite3.so.0 libssl.so.1.1 libstdc++.so.6 libstrongswan-kernel-netlink.so libstrongswan.so.0 libsys_util.so libtls.so.0 libtmctl.so libubox.so libusb-0.1.so.4 libusb-1.0.so.0 libutil.so.1 libuuid.so.1 libvici.so.0 libvlanctl.so libvorbis.so.0 libwebapi.so libwlc_nt_client.so libwlcsm.so libwpa_client.so libws.so libxml2.so.2 libxtables.so.7 libz.so.1 libzebra.so.0 libnss_dns.so.2 libnss_files.so.2
Makefile:221: 32-bit libraries: libbcm_crc.so

###################################################
Installing C run-time library...
######### All 32-bit required shared libs #########
libFLAC.so.8 libamas-utils.so libarcher.so libasc.so libasuslog.so libatalk.so.6 libavahi-common.so.3 libavahi-core.so.7 libavcodec.so.58 libavformat.so.58 libavutil.so.56 libbcm.so libbcm_boardctl.so libbcm_crc.so libbcm_flashutil.so libbcm_util.so libbcmmcast.so libbcmtm.so libblkid.so.1 libblobmsg_json.so libblogctl.so libbridgeutil.so libbwdpi.so libbwdpi_sql.so libc.so.6 libcap-ng.so.0 libceshared.so libcfgmnt.so libcharon.so.0 libcodb.so libcom_err.so.2 libconn_diag.so libcrypt.so.1 libcrypto.so.1.1 libcurl.so.4 libdaemon.so.0 libdb-4.8.so libdisk.so libdl.so.2 libe2p.so.2 libebt_arp.so libebt_arpreply.so libebt_ftos.so libebt_ip.so libebt_limit.so libebt_log.so libebt_mark.so libebt_pkttype.so libebt_redirect.so libebt_skiplog.so libebt_standard.so libebt_stp.so libebt_time.so libebt_vlan.so libebt_wmm_mark.so libebtable_broute.so libebtable_filter.so libebtable_nat.so libebtc.so libethctl.so libethswctl.so libev.so.4 libevent-2.0.so.5 libexif.so.12 libexpat.so.1 libext2fs.so.2 libfcctl.so libgcc_s.so.1 libgcrypt.so.11 libgdbm.so.3 libgen_util.so libgpg-error.so.0 libiconv.so.2 libid3tag.so.0 libip4tc.so.0 libip6tc.so.0 libiqctl.so libjansson.so.4 libjpeg.so libjson-c.so.2 libletsencrypt.so liblightsql.so liblldpctl.so.4 liblz4.so.1 liblzo2.so.2 libm.so.6 libmnl.so.0 libmssl.so libncurses.so.6 libneon.so.27 libnetfilter_conntrack.so.3 libnetsnmp.so.35 libnetsnmpagent.so.35 libnetsnmpmibs.so.35 libnfnetlink.so.0 libnl-3.so.200 libnl-genl-3.so.200 libnl-nf-3.so.200 libnl-route-3.so.200 libnsl.so.1 libnt.so libnvram.so libogg.so.0 libovpn.so libpam.so.2 libpasswd.so libpcre.so.1 libpng16.so.16 libptcsrv.so libpthread.so.0 libpwrctl.so libqrencode.so.4 libresolv.so.2 librt.so.1 libshared.so libshn_pctrl.so libshn_utils.so libsmartsync_api.so libsmbclient.so.0 libsqlite3.so.0 libssl.so.1.1 libstdc++.so.6 libstrongswan-kernel-netlink.so libstrongswan.so.0 libsys_util.so libtls.so.0 libtmctl.so libubox.so libusb-0.1.so.4 libusb-1.0.so.0 libutil.so.1 libuuid.so.1 libvici.so.0 libvlanctl.so libvorbis.so.0 libwebapi.so libwlc_nt_client.so libwlcsm.so libwpa_client.so libws.so libxml2.so.2 libxtables.so.7 libz.so.1 libzebra.so.0 libnss_dns.so.2 libnss_files.so.2
###################################################

[... cut ...]
make[3]: *** No rule to make target '/build/release/src-rt-5.04axhnd.675x/hostTools/prelink/src/execstack', needed by 'build_execstack'.  Stop.
make[3]: Leaving directory '/build/release/src-rt-5.04axhnd.675x/hostTools'
make[2]: *** [build/Makefile:822: full_buildimage] Error 2
make[2]: Leaving directory '/build/release/src-rt-5.04axhnd.675x'
make[1]: *** [Makefile:5999: bin] Error 2
make[1]: Leaving directory '/build/release/src-rt-5.04axhnd.675x'
make: *** [Makefile:6223: rt-ax82u_v2] Error 2
`

gnuton, 2023-10-27:
```
full_buildimage buildimage:
ifeq ($(BUILD_DISABLE_EXEC_STACK),y)
ifneq ($(execstack_exec),)
        @echo no need to build execstack $(execstack_exec)
else
        make -C $(HOSTTOOLS_DIR) build_execstack;
endif
endif
```

gnuton, 2023-11-11:
Fixed https://github.com/gnuton/asuswrt-merlin.ng/issues/475

closed 2023-11-11 by gnuton

Moshe1010, 2023-12-31:
Not sure if it's just me, but when I try to upgrade to 3004.388.5_0, the version doesn't change after uploading the following file:
RT-AX82U_3004_388.5_0-gnuton1_puresqubi.w

I try to upgrade it as a Mesh unit. 

The other Mesh I have (AX5400) works great, but 82U_V2 doesn't want to take the newest version. 

My setup is basically AX1100 works as the main unit, then AX5400 and 82U_V2 are operating as Mesh units.

Thanks