gnuton/asuswrt-merlin.ng #475: Unable to flash on RT-AX82U v2

ZorianNL, 2023-10-04:
**Router Model Affected**
Models: (eg: DSL-AX82U/RT-AX82U/...)
RT-AX82U v2

**Firmware Version Affected**
3004.388.4_0-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
I cannot upload the firmware file to my router. All other files have the .w extension but this file does not, and for some reason the Beta 1 file is shown under Beta 2 aswell. I can select the file just fine on the firmware update page, but after the "flash" is done nothing has happened.

**To Reproduce**
Steps to reproduce the behavior:
Update firmware, nothing happens because probably wrong file extension/file.

**Expected behavior**
Firmware flashed successfully.

gnuton, 2023-10-04:
thanks for reporting this. The problem is not the extension, but I think the problem is something nasty that happens during linking phase of some executables in the last stages of the build. I just need to rebuild the code and check.

gnuton, 2023-10-04:
https://github.com/gnuton/asuswrt-merlin.ng/issues/474 is linked to this issue

ZorianNL, 2023-10-04:
Using the file from beta1 worked by the way. It's flashed, however the 5Ghz doesn't seem to work correctly. I can see the network if I split the frequencies, but using Smart Connect it's just 2.4Ghz and 5Ghz is never used. Hopefully I can add that issue to this report.

ZorianNL, 2023-10-04:
To add, using the file from beta 1 instead of beta 2 worked, it has flashed but has a few quirks (see other issue reports, such as wifi insight giving the 404).

gnuton, 2023-10-04:
please file 1 bug per issue

gnuton, 2023-10-05:
Here the syslog after uploading a broken image
[syslog.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/12822615/syslog.txt)

gnuton, 2023-10-05:
Checked dhd prebuilts and they are fine
./release/src/router/rom/dhd
./release/src-rt-5.04axhnd.675x/bcmdrivers/broadcom/net/wl/impl87/sys/src/dhd/exe/prebuilt/dhd
./release/src-rt-5.04axhnd.675x/bcmdrivers/broadcom/net/wl/impl87/sys/src/dhd/exe/prebuilt/dhd

ZorianNL, 2023-10-05:
Yes, the build from beta 1 flashed for me, but the one that was previously linked in beta 2 did not. Something is definitely different between the 2. Looks like it happens pretty early on aswell.

gnuton, 2023-10-05:
from the build looks these lines look suspicious too
```/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32/usr/bin/arm-buildroot-linux-gnueabi-strip: unable to copy file '/build/release/src-rt-5.04axhnd.675x/targets/96750GW/fs/sbin/wpsaide'; reason: Permission denied
/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32/usr/bin/arm-buildroot-linux-gnueabi-strip: unable to copy file '/build/release/src-rt-5.04axhnd.675x/targets/96750GW/fs/sbin/write_3g_ppp_conf'; reason: Permission denied
/opt/toolchains/crosstools-arm-gcc-9.2-linux-4.19-glibc-2.30-binutils-2.32/usr/bin/arm-buildroot-linux-gnueabi-strip:/build/release/src-rt-5.04axhnd.675x/targets/96750GW/fs/usr/bin/net-snmp-config: file format not recognized
```

gnuton, 2023-10-05:
Complete [build logs from circle ci](https://circleci-tasks-prod.s3.us-east-1.amazonaws.com/storage/artifacts/bf22d1be-9a12-46b1-b4a9-afb85adb0c62/62fe19af-1ba2-4a43-a398-476255df397f/0/workspace/release/rt-ax82u_v2-build.log?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=ASIAQVFQINEOGLRPOYOM%2F20231005%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231005T200855Z&X-Amz-Expires=60&X-Amz-Security-Token=IQoJb3JpZ2luX2VjEF0aCXVzLWVhc3QtMSJHMEUCIQCaylfDwcVOxl95dwFIkNShr5H5qGGiCUaCyVCK2xSWWAIgdsVsljExHxooONaPRZgZ8kn%2BDsET5IweOzOJ1Xfh3BYqqwIIZRADGgwwNDU0NjY4MDY1NTYiDK2LwFF499m%2FBD7uASqIAuGDj4rTJ%2FUGRieveztlLjBGLGETl2ArA%2Bme4fb7Nr%2BvfDvVyynt9wFFMY9O8FhBQD3bQtQ88U3zylJuSuXe38q83eQ5K2KnNLptnR8YrDw6F2J8rkTxc5AW19yErbXNvMpHcXlf4RD%2BUB5qxlN6ghF3U7N9HJbag%2BFL%2Ftk0qcmc6vtEytN6mjtqMApjZ%2BkBnYyAj8fW4JCLjod7z8tZ%2Bh3vxHpYzwUsZCpuYBrpjC4FdzhhcR8eYE0GbBQztNdC4qW86vEBZMnv08Svx7%2FgvatPN1t7wnBJ%2BOJpwHsgBWsy1ZzCK9tQ6sO7P0D%2BG9WYkb46vsHp9odKV2qlGQ8iQSYcaDo%2F0MbWNDDervyoBjqdAUw0SNYES%2Bv6O3ng4Ev%2Bu9PkIHEmX%2BPvaMVG6Yel7SMKSufVtVSYGCZHNWyxpmV2hTuX%2FatrD9N0vx9mRXcXEP0MuzQWBj9npHvGJl4gpGRkc%2FhVVG7sRIleoVGXpkJRiTdUtVa9cmIfxhh89PIOeZbi1iXdjzf5gV9llHSmnEfqYbnVWNm8vXN1UFxxAY9o6UEl1OyoWvlNc7a1UYY%3D&X-Amz-SignedHeaders=host&x-id=GetObject&X-Amz-Signature=7565c0c7b03d9ba690b1a5ca98e75901b16ba9d2e68acf80c836ba7a4a663712)

ZorianNL, 2023-10-05:
Those definitely will cause problems yeah.

gnuton, 2023-10-09:
I checked once again the GPL that I got from ASUS and it's broken
The build fails with

CONFIG_BCM_BOOTSTATE_IMPL=1
cp ../../bcmdrivers/broadcom/char/archer/bcm963178/archer.o ../../bcmdrivers/broadcom/char/archer/bcm963178/archer_dep.o
cp: cannot stat '../../bcmdrivers/broadcom/char/archer/bcm963178/archer.o': No such file or directory
make[6]: *** [../../bcmdrivers/broadcom/char/archer/bcm963178/Makefile:6: ../../bcmdrivers/broadcom/char/archer/bcm963178/archer_dep.o] Error 1
make[5]: *** [scripts/Makefile.build:565: ../../bcmdrivers/broadcom/char/archer/bcm963178] Error 2
make[4]: *** [Makefile:1081: ../../bcmdrivers] Error 2
make[4]: Leaving directory '/build/asuswrt/release/src-rt-5.04axhnd.675x/kernel/linux-4.19'
make[3]: *** [Bcmkernel.mk:83: default] Error 2
make[3]: Leaving directory '/build/asuswrt/release/src-rt-5.04axhnd.675x/build'
make[2]: *** [build/Makefile:509: kernelbuild] Error 2
make[2]: Leaving directory '/build/asuswrt/release/src-rt-5.04axhnd.675x'
make[1]: *** [Makefile:5952: bin] Error 2
make[1]: Leaving directory '/build/asuswrt/release/src-rt-5.04axhnd.675x'
make: *** [Makefile:6173: rt-ax82u_v2] Error 2**

as it was failing for me before my fix. Moving the archer.o from hnd_extras folder to the SDK folder fixes the problem and the built stock image looks to flash and work correctly.

According to Eric the problem should be in the archer driver.

Johankp, 2023-10-10:
This also happens to my v1 of RT AX82U. It fukked up my whole wifi system at home. I have a 82u as a nod to GT-AX6000 router in a wifi mesh. I tried downgrading to earlier version but no success. Not even with official fw 🤔

TelecomDoctor, 2023-10-12:
Not clear from reading through here what the status is on Beta 2 for the 82U V2?  Is this any closer to resolution or is the stock GPL code from Asus also in trouble? I don't want to F*** my router but I wanted to put it into production, like yesterday....;-)

gnuton, 2023-10-13:
@TelecomDoctor the RT-AX82U_V2 is still unstable so it remains in beta. I will push a new beta when i get this fixed.

ZorianNL, 2023-10-13:
> Not clear from reading through here what the status is on Beta 2 for the 82U V2? Is this any closer to resolution or is the stock GPL code from Asus also in trouble? I don't want to F*** my router but I wanted to put it into production, like yesterday....;-)

Beta 1 works for me on my v2.

TelecomDoctor, 2023-10-13:
> > Not clear from reading through here what the status is on Beta 2 for the 82U V2? Is this any closer to resolution or is the stock GPL code from Asus also in trouble? I don't want to F*** my router but I wanted to put it into production, like yesterday....;-)
> 
> Beta 1 works for me on my v2.

Yes thanks for chiming in. I did/do have the B1 on my V2.  There is a BUG report #476 open on it.

TelecomDoctor, 2023-10-21:
> @TelecomDoctor the RT-AX82U_V2 is still unstable so it remains in beta. I will push a new beta when i get this fixed.

Hi there - any updated status for the Beta 2 for this router?  Thanks

gnuton, 2023-10-26:
@TelecomDoctor the RT-AX82U_v2 comes with a new SDK and something is not right. I am still investigating.. 
I have not given up yet.

ZorianNL, 2023-11-03:
Found anything new on this yet? So far the Beta 1 is still running fine without a hitch for me.

gnuton, 2023-11-03:
Fixed a few things, but the main problem is still there and to me it still feel like it's in the building system of the new SDK.  I think I will try to release a new image soon.

sadoneli, 2023-11-09:
To fix this issue for RT-AX82U_V2, just do as below:

```bash
cd release/src-rt-5.04axhnd.675x/bcmdrivers/broadcom/net/wl
ln -sf impl87 bcm963178
```

then you can run your build process

gnuton, 2023-11-10:
thanks @sadoneli testing... the symlink is actually missing.. but I thought it was created by the build sistem
from the logs:
```
...
linking broadcom/net/wl/bcm963178 => impl87
...
```

gnuton, 2023-11-11:
The problem was indded related to the missing symlink. 
d7a469e87a5e9fac71059cc22a78ca5af43d9297

I had to fix also a few smaller stuff
42128bac52c7f41ca3f3cb077b7129b5777bfece - use local execstack
ffc7faa360d16eb9fa3c989af17a7c67f190432d - fix libcreduction Makefile which may fail if some dependencies are in "NEEDED"
4c655c955f248d93ed1da13dec162f0686700d5a - do not execstack and libelf from GPLs

gnuton, 2023-11-11:
you can find the stable release for the current stable

closed 2023-11-11 by gnuton