gnuton/asuswrt-merlin.ng #476: RT-AX82u v2 3004.388.4_0-knuton0_beta1 missing WiFi_Insight.asp file

TelecomDoctor, 2023-10-04:
**Router Model Affected**
RT-AX82u v2 

**Firmware Version Affected**
3004.388.4_0-knuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Not known

**Describe the bug**
Getting a "404 file not found for WiFi_Insight.asp

Possible missing 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Network Tools'
2. Click on 'WiFi Radar'
3. See 404 error

Imastupidsalmon, 2024-01-04:
This issue is also occuring on the rt-ax5400. 3004.388.6_alpha1

gnuton, 2024-02-13:
this should be fixed, do you confirm?

Imastupidsalmon, 2024-02-13:
@gnuton I can confirm that in the RT-AX5400 it is indeed fixed in the latest build. Nice. 👍 Thank you!
Super cool!!
[image]
[image]

gnuton, 2024-02-14:
Awesome thanks I will close this issue.

closed 2024-02-14 by gnuton