gnuton/asuswrt-merlin.ng #477: Missing firmware manual upload section

DaveEzrakhovich, 2023-10-06:
**Router Model Affected**
Models: DSL-AX82U / DSL-AX5400

**Firmware Version Affected**
388.1-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
After the new version was released, I was notified that I could update it.
I went to the firmware section in the router UI portal, but the upload options were missing. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration'
2. Click on 'Firmware Upgrade' section at the top

**Expected behavior**
Option to upload the firmware. 

**Screenshots**
[image]

**Notes**
I've been using the gunton builds for several years without any issues, this specific device was upgraded several times in the past.

gnuton, 2023-10-06:
there could be a regression which was already fixed once https://github.com/gnuton/asuswrt-merlin.ng/issues/144
I will check. In any case thisi for an old firmware. You should upgrade to the last one.
To make show the upgrade option you have to trick chrome

DaveEzrakhovich, 2023-11-03:
Thanks, @gnuton. I've upgraded to the newest version using browser debug
But the problem still exists in the 3004.388.4_0-gnuton1 version 

Anyway, if other users have the same issue, here is the browser quick fix that will allow you to upgrade 

1. Open the router admin page + login 
2. Go to administration -> firmware upgrade 
3. Open dev tools (F12 or right-click and click inspect) 
4. Go to the sources tab and add a breakpoint on the line with "var exist_update = 0;" 
5. refresh the page
6. Go to the console tab and run the following: no_fw_manual_support = 0; exist_update = 1; no_update_support = 0; 
7. Press resume (F8)
8. Upload the firmware and good luck

Jsonmess, 2024-04-18:
i try the last version 3004.388.6_0-gnuton1 in dsl_5400/ds_82u,this issue is still exit； @gnuton

gnuton, 2024-04-18:
Do you have a sort of branded router?

dave-aspecto, 2024-04-18:
Nope native
The issue happened on a couple of versions, but as the workaround works I didn't say anything 😅

DaveEzrakhovich, 2024-04-18:
> Do you have a sort of branded router?

Nope native
The issue happened on a couple of versions, but as the workaround works I didn't say anything 😅

DaveEzrakhovich, 2026-10-12:
@gnuton seem to be good
sorry for the late respsone