gnuton/asuswrt-merlin.ng #478: Random crash RT-AX92U

megosugit, 2023-10-08:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Sorry, I don't know.

**Describe the bug**
From time to time, I don't have access to Internet anymore, and I can't even reach other devices connected on the same Wifi.
If I wait a few minutes, it all comes back.

**To Reproduce**
Steps to reproduce the behavior:
Happens randomly

**Expected behavior**
Router should be more stable.

**Screenshots**
I can provide the System logs, when it happens I can see it goes into crashlog and the date goes back to 7 may during initialization.

It looks like this, should I poste the whole systemlogs ?

megosugit, 2023-10-08:
not sure if it is related, but I have a lot of wl0 errors and resets:

> Oct  8 22:01:20 kernel: wl0: PSM microcode watchdog fired at 150157 (seconds). Resetting.
Oct  8 22:01:20 kernel: wl0: reason = psm watchdog at 150157 seconds. corerev 42 ucode revision 1570.181 features 0x0426
Oct  8 22:01:20 kernel: psmdebug 0x00ff8c64 phydebug 0x71 macctl 0x84160403 maccmd 0x4
Oct  8 22:01:20 kernel: psm_brc 0x0242 psm_brc_1 0x003c M_UCODE_DBGST 0x2
Oct  8 22:01:20 kernel: brwk_0 0x0094 brwk_1 0x0002 brwk_2 0x0030 brwk_3 0x7123 brwk_4 0x0000
Oct  8 22:01:20 kernel: ifsstat      0x9ea0 ifsstat1     0x1f0  txectl       0x4001 txestat      0x401  
Oct  8 22:01:20 kernel: rxestat1     0x3b22 rxestat2     0x4a00 rcv_frmcnt   0x0    rxe_rxcnt    0x3b2  
Oct  8 22:01:20 kernel: wepctl       0x0    
Oct  8 22:01:20 kernel: aqmfifordy   0x0    wepstat      0x0    wep_ivloc    0x1e   wep_psdulen  0xe    
Oct  8 22:01:20 kernel: PC :
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
Oct  8 22:01:20 kernel: R0: 0xff8c26   0xc998e1   0xff98e8   0xec9470  
Oct  8 22:01:20 kernel: phydebug :
Oct  8 22:01:20 kernel: 0x40       0x41       0x40       0x40      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: 0x31       0x31       0x31       0x31      
Oct  8 22:01:20 kernel: 0x40       0x40       0x40       0x40      
Oct  8 22:01:20 kernel: pktproc      0x101  pktproc      0x101  pktproc      0x101  pktproc      0x101  
Oct  8 22:01:20 kernel: psm stack_status : 0x10
Oct  8 22:01:20 kernel: psm stack_entries:
Oct  8 22:01:20 kernel: 0x00e6
Oct  8 22:01:20 kernel: 0x1470
Oct  8 22:01:20 kernel: 0x11d3
Oct  8 22:01:20 kernel: 0x11ee
Oct  8 22:01:20 kernel: 0x0d7d
Oct  8 22:01:20 kernel: 0x1eff
Oct  8 22:01:20 kernel: 0x1d57
Oct  8 22:01:20 kernel: 0x1fff
Oct  8 22:01:20 kernel: wl0: fatal error, reinitializing, total count of reinit's[684]

gnuton, 2023-10-09:
This is not a crash, these are just debug messages. A  kernel crash comes with a crashlog and a stacktrace.
I would suggest you to update to latest beta2 which may have updated wireless drivers or you may wanna to reset to factory to start with fresh settings and see if there is anything of your settings which would trigger this behaviour in the wireless driver

megosugit, 2023-10-09:
Hi,

I posted some bits of the crashlog messages also in my first message, and then I posted about the repeated wl0 errors in my second message as I was wondering if they could be related and cause the full crash.

I have inserted the full crashlog below, are you able to see a clear culprit ?

I see for instance:
May  5 07:05:06 crashlog: <4>aaews triggered out of memory codition (oom killer not called): gfp_mask=0x201da, order=0, oom_score_adj=0
May  5 07:05:06 crashlog: <0>Kernel panic - not syncing: Reboot due to persistent out of memory codition..

<details><summary> full crashlog</summary>
<p>

May  5 07:05:06 crashlog: LOG
May  5 07:05:06 crashlog: <6>[ 1535]     0  1535     1251       82       4       2        0             0 fsmd
May  5 07:05:06 crashlog: <6>[ 1657]     0  1657     7838      293      18       2        0             0 charon
May  5 07:05:06 crashlog: <6>[ 1710]     0  1710      630       51       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 1711]     0  1711      630       51       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 1738]     0  1738      630       49       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 1746]     0  1746      630       51       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 1747]     0  1747      630       51       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 1990]     0  1990     4636      351      10       2        0             0 dcd
May  5 07:05:06 crashlog: <6>[ 2017]     0  2017     3959      789      11       2        0             0 wred
May  5 07:05:06 crashlog: <6>[ 2020]     0  2020     3168      133      10       2        0             0 bwdpi_wred_aliv
May  5 07:05:06 crashlog: <6>[ 2025]     0  2025      431       15       5       2        0             0 tcd
May  5 07:05:06 crashlog: <6>[ 3245]     0  3245     2537       76       9       2        0             0 smbd
May  5 07:05:06 crashlog: <6>[ 3246]     0  3246     2467       54       8       2        0             0 nmbd
May  5 07:05:06 crashlog: <6>[ 3248]     0  3248     2397       31       7       2        0             0 nmbd
May  5 07:05:06 crashlog: <6>[ 3249]     0  3249      468       37       5       2        0             0 wsdd2
May  5 07:05:06 crashlog: <6>[ 3250]     0  3250      630       49       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 3650]     0  3650      630       51       5       2        0             0 dropbear
May  5 07:05:06 crashlog: <6>[ 4383] 65534  4383      693      107       3       2        0             0 dnsmasq
May  5 07:05:06 crashlog: <6>[ 4385]     0  4385      661       80       3       2        0             0 dnsmasq
May  5 07:05:06 crashlog: <6>[ 4486]     0  4486      487       23       5       2        0             0 mcpd
May  5 07:05:06 crashlog: <6>[ 4599]     0  4599     3168      125      10       2        0             0 usbled
May  5 07:05:06 crashlog: <6>[ 4600]     0  4600     2016       61       7       2        0             0 usbmuxd
May  5 07:05:06 crashlog: <6>[ 4877]     0  4877      944       36       4       2        0             0 ntp
May  5 07:05:06 kernel: klogd started: BusyBox v1.25.1 (2023-06-01 06:04:36 UTC)
May  5 07:05:06 crashlog: <6>[ 4904]     0  4904     1685      404       8       2        0             0 stubby
May  5 07:05:06 crashlog: <6>[ 4906]     0  4906     3168      127      10       2        0             0 disk_monitor
May  5 07:05:06 kernel: Linux version 4.1.51 (docker@a3569e6e5523) (gcc version 5.5.0 (Buildroot 2017.11.1) ) #2 SMP PREEMPT Thu Jun 1 06:11:53 UTC 2023
May  5 07:05:06 crashlog: <6>[ 4973]     0  4973     4448      145      12       2        0             0 amas_portstatus
May  5 07:05:06 kernel: CPU: AArch64 Processor [420f1000] revision 0
May  5 07:05:06 crashlog: <6>[ 5989]     0  5989     4894      426      13       2        0             0 aaews
May  5 07:05:06 crashlog: <6>[ 6282]     0  6282      630       49       5       2        0             0 dropbear
May  5 07:05:06 kernel: Kernel command line: coherent_pool=4M cpuidle_sysfs_switch pci=pcie_bus_safe rootwait
May  5 07:05:06 crashlog: <6>[ 6292]     0  6292      653       65       5       2        0             0 dropbear
May  5 07:05:06 kernel: Virtual kernel memory layout:
May  5 07:05:06 kernel:     vmalloc : 0xffffff8000000000 - 0xffffffbdffff0000   (   247 GB)
May  5 07:05:06 kernel:     vmemmap : 0xffffffbe00000000 - 0xffffffbfc0000000   (     7 GB maximum)
May  5 07:05:06 kernel:               0xffffffbe00000000 - 0xffffffbe00700000   (     7 MB actual)
May  5 07:05:06 crashlog: <6>[31538]     0 31538      630       53       5       2        0             0 dropbear
May  5 07:05:06 kernel:     fixed   : 0xffffffbffabfd000 - 0xffffffbffac00000   (    12 KB)
May  5 07:05:06 kernel:     PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
May  5 07:05:06 kernel:     modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
May  5 07:05:06 kernel:     memory  : 0xffffffc000000000 - 0xffffffc020000000   (   512 MB)
May  5 07:05:06 kernel:       .init : 0xffffffc0006ce000 - 0xffffffc000707000   (   228 KB)
May  5 07:05:06 crashlog: <6>[27749]     0 27749      630       45       5       2        0             0 dropbear
May  5 07:05:06 kernel:       .text : 0xffffffc000080000 - 0xffffffc0006cd204   (  6453 KB)
[... cut ...]
May  5 07:05:59 rc_service: wanduck 1152:notify_rc start_ntpd
May  5 07:05:59 rc_service: waitting "stop_ntpd" via wanduck ...
May  5 07:05:59 ntpd: Stopped ntpd
May  5 07:06:00 ntpd: Started ntpd
Oct  8 22:23:29 ntpd: Initial clock set
Oct  8 22:23:29 rc_service: ntpd_synced 5462:notify_rc restart_stubby
Oct  8 22:23:29 rc_service: ntpd_synced 5462:notify_rc restart_diskmon
Oct  8 22:23:29 rc_service: waitting "restart_stubby" via ntpd_synced ...
Oct  8 22:23:30 disk_monitor: Finish

</p>
</details>

gnuton, 2023-10-09:
I may wrong  but this seems to me the most important part of this log..
the issues seems that aaews generated a memory issues. 
I do see the kernel crash here.
Can you please test the beta2 and report back if you stil see this issue?

`May 5 07:05:06 crashlog: <6>[22064] 0 22064 856 23 5 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22065] 0 22065 3168 122 10 2 0 0 ATE
May 5 07:05:06 crashlog: <6>[22069] 0 22069 630 46 5 2 0 0 dropbear
May 5 07:05:06 crashlog: <6>[22070] 0 22070 630 46 5 2 0 0 dropbear
May 5 07:05:06 crashlog: <6>[22073] 0 22073 856 23 4 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22074] 0 22074 1361 59 7 2 0 0 AiProtectionMon
May 5 07:05:06 crashlog: <6>[22076] 0 22076 630 46 5 2 0 0 dropbear
May 5 07:05:06 crashlog: <6>[22089] 0 22089 856 23 5 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22096] 0 22096 856 23 6 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22101] 0 22101 598 31 5 2 0 0 dropbear
May 5 07:05:06 crashlog: <6>[22104] 0 22104 3168 123 10 2 0 0 PS_pod
May 5 07:05:06 crashlog: <6>[22106] 0 22106 856 25 6 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22113] 0 22113 995 39 5 2 0 0 wl
May 5 07:05:06 crashlog: <6>[22115] 0 22115 598 35 5 2 0 0 dropbear
May 5 07:05:06 crashlog: <6>[22116] 0 22116 995 39 4 2 0 0 wl
May 5 07:05:06 crashlog: <6>[22118] 0 22118 486 14 4 2 0 0 nvram
May 5 07:05:06 crashlog: <6>[22119] 0 22119 263 8 3 2 0 0 sh
May 5 07:05:06 crashlog: <6>[22120] 0 22120 661 80 3 2 0 0 dnsmasq
May 5 07:05:06 crashlog: <6>[22121] 0 22121 1361 59 6 2 0 0 AiProtectionMon
May 5 07:05:06 crashlog: <4>
May 5 07:05:06 crashlog: <4>
May 5 07:05:06 crashlog: <4>aaews triggered out of memory codition (oom killer not called): gfp_mask=0x201da, order=0, oom_score_adj=0
May 5 07:05:06 crashlog: <4>
May 5 07:05:06 crashlog: <4>CPU: 0 PID: 6100 Comm: aaews Tainted: P O 4.1.51 https://github.com/gnuton/asuswrt-merlin.ng/pull/2
May 5 07:05:06 crashlog: <4>Hardware name: Broadcom-v8A (DT)
May 5 07:05:06 crashlog: <0>Call trace:
May 5 07:05:06 crashlog: <4>[] dump_backtrace+0x0/0x150
May 5 07:05:06 crashlog: <4>[] show_stack+0x14/0x20
May 5 07:05:06 crashlog: <4>[] dump_stack+0x90/0xb0
May 5 07:05:06 crashlog: <4>[] __out_of_memory.isra.6+0xa4/0x148
May 5 07:05:06 crashlog: <4>[] out_of_memory+0x48/0x68
May 5 07:05:06 crashlog: <4>[] __alloc_pages_nodemask+0x610/0x6a0
May 5 07:05:06 crashlog: <4>[] filemap_fault+0x15c/0x3c0
May 5 07:05:06 crashlog: <4>[] __do_fault+0x34/0x90
May 5 07:05:06 crashlog: <4>[] handle_mm_fault+0x9e4/0x1120
May 5 07:05:06 crashlog: <4>[] do_page_fault+0x10c/0x268
May 5 07:05:06 crashlog: <4>[] do_mem_abort+0x3c/0x98
May 5 07:05:06 crashlog: <4>Exception stack(0xffffffc006917e20 to 0xffffffc006917f50)
May 5 07:05:06 crashlog: <4>7e20: 00400000 00000000 00000000 00000000 ffffffff ffffffff f6a5a438 00000000
May 5 07:05:06 crashlog: <4>7e40: 06917e50 ffffffc0 000ee064 ffffffc0 00000000 00000000 000845f0 ffffffc0
May 5 07:05:06 crashlog: <4>7e60: 00400000 00000000 00000000 00000000 ffffffff ffffffff f6cc250c 00000000
May 5 07:05:06 crashlog: <4>7e80: 06917ea0 ffffffc0 000871a8 ffffffc0 00400008 00000000 000845f0 ffffffc0
May 5 07:05:06 crashlog: <4>7ea0: 00000000 00000000 000844d4 ffffffc0 00400000 00000000 019a6298 00000000
May 5 07:05:06 crashlog: <4>7ec0: 0000006e 00000000 00000000 00000000 00000002 00000000 000017d4 00000000
May 5 07:05:06 crashlog: <4>7ee0: f6ae9300 00000000 f66fed20 00000000 f6ae9260 00000000 f66fed28 00000000
May 5 07:05:06 crashlog: <4>7f00: ff9c61fa 00000000 ff9c6200 00000000 00000000 00000000 f66fed18 00000000
May 5 07:05:06 crashlog: <4>7f20: 00000000 00000000 f66fed10 00000000 f6cc2674 00000000 00000000 00000000
May 5 07:05:06 crashlog: <4>7f40: 00000000 00000000 00000000 00000000
May 5 07:05:06 crashlog: <4>Mem-Info:
May 5 07:05:06 crashlog: <4>active_anon:22525 inactive_anon:1367 isolated_anon:0
May 5 07:05:06 crashlog: <4> active_file:34 inactive_file:27 isolated_file:15
May 5 07:05:06 crashlog: <4> unevictable:0 dirty:0 writeback:0 unstable:0
May 5 07:05:06 crashlog: <4> slab_reclaimable:609 slab_unreclaimable:42991
May 5 07:05:06 crashlog: <4> mapped:158 shmem:1546 pagetables:1833 bounce:0
May 5 07:05:06 crashlog: <4> free:5759 free_pcp:224 free_cma:0
May 5 07:05:06 crashlog: <4>DMA free:22924kB min:20480kB low:25600kB high:30720kB active_anon:90100kB inactive_anon:5468kB active_file:52kB inactive_file:168kB unevictable:0kB isolated(anon):0kB isolated(file):4kB present:464896kB managed:408852kB mlocked:0kB dirty:0kB writeback:
May 5 07:05:06 crashlog: 0kB mapped:708kB shmem:6184kB slab_reclaimable:2436kB slab_unreclaimable:171964kB kernel_stack:6784kB pagetables:7332kB unstable:0kB bounce:0kB free_pcp:964kB local_pcp:424kB free_cma:0kB writeback_tmp:0kB pages_scanned:1516 all_unreclaimable? yes
May 5 07:05:06 crashlog: <4>lowmem_reserve[]: 0 0 0
May 5 07:05:06 crashlog: <4>DMA: 5924kB (UEM) 1198kB (UEM) 16516kB (UEM) 17732kB (UEM) 4364kB (UEM) 3128kB (E) 0256kB 0512kB 01024kB 02048kB 2*4096kB (R) = 22952kB
May 5 07:05:06 crashlog: <4>1633 total pagecache pages
May 5 07:05:06 crashlog: <4>0 pages in swap cache
May 5 07:05:06 crashlog: <4>Swap cache stats: add 0, delete 0, find 0/0
May 5 07:05:06 crashlog: <4>Free swap = 0kB
May 5 07:05:06 crashlog: <4>Total swap = 0kB
May 5 07:05:06 crashlog: <4>116224 pages RAM
May 5 07:05:06 crashlog: <4>0 pages HighMem/MovableOnly
May 5 07:05:06 crashlog: <4>14011 pages reserved`

megosugit, 2023-10-09:
I usually stick to stable release. I have read on this forum that some users bricked their device with a beta version.

any ETA for stable release ?

BoostOver, 2023-10-09:
> I usually stick to stable release. I have read on this forum that some users bricked their device with a beta version.
> 
> any ETA for stable release ?

Flash only official firmware and you are safe….+ or -

BoostOver, 2023-10-09:
> if it is uncertain, unfortunately if it bricks I don't have the money to buy a new router :(

https://www.asus.com/support/FAQ/1000814/

megosugit, 2023-10-09:
so it seems impossible to brick a router with Gnuton release, good to know.

is there an ETA for stable ?

megosugit, 2023-10-10:
I got another crash last night.

Since I need a stable Wifi (because of work and home automation), I couldn't stay like this.
I installed beta2 on both meshed routers hoping it would fix the issue. I'll report the results.

When I bought these routers, I quickly moved from stock to merlin FW hoping to get rocksteady wifi, but since I bought these 2 routers (2*RT-AX92U), I had really mixed results and they are far from being plug and forget. Is it an Asus issue ? Did I make a mistake choosing this brand or this model ?

One of my recurring issues, is that some devices (different brands) randomly lose their Internet connectivity. I can ping them and control them on the LAN but they don't report back to their online servers (Xiaomi, Yeelight and Broadcom for instance).
Even if I disconnect them and plug them back, same result.
I found 2 different ways to temporarily fix it:
1/ I can leave them unplugged for several minutes (5min at least), then plug them back.
2/ if I changed their DHCP IP reservation, then they come back alive (but since it restarts the wifi, sometimes I lose another device)
Both options work, until next time :(

Any thoughts about this ? It drives me crazy 🤪

gnuton, 2023-10-12:
@megosugit let's not add anything to the ticket which doesn't belong to the crash/this bug.
Please use the chat or forum for these questions there are more people there and you will get a better answer. Quite sure this happens not only on RT-AX92U.

megosugit, 2023-10-21:
Hello,
I was going to say it was stable with beta2, unfortunately it happened again tonight.
It is a bit more stable I would say, but still rebooting unexpectedly.

Same type of error in todays crash log, with the same spelling mistakes:
May  5 07:05:04 crashlog: <4>debug_monitor triggered out of memory codition (oom killer not called): gfp_mask=0x201da, order=0, oom_score_adj=0
May  5 07:05:04 crashlog: <0>Kernel panic - not syncing: Reboot due to persistent out of memory codition..

is there something else I could try ? beta3 maybe or stable ? :D

gnuton, 2023-10-25:
@megosugit RT-AX92U won't change so much between beta2 and stable.
have you tested stock firmware?

megosugit, 2023-10-25:
I'll try stable, then if it continues, I'll need to go back to stock to check and if it continues, I might even break the mesh, which is not ideal... but I need to find a solution.
Do you have an ETA for stable ?

Will I need to redo all my settings going to stock (that's the worst time consuming for me) ?

megosugit, 2023-10-26:
I have flashed stable on main router and on mesh, thanks !!
I will report if something happens.

If I need to go back to stock, would I need to redo all my settings (it's extremely time consuming) ?

gnuton, 2023-10-26:
you could keep the merlin settings, but you may notice issues.
When you see problems the best thing to do is to factory reset and add setting little by little so that you understand wha has casued the issue.

megosugit, 2023-10-27:
unfortunately, it is still the same and it was quick to happen this time.
May  5 07:05:04 crashlog: <4>dnsmasq triggered out of memory codition (oom killer not called): gfp_mask=0xd0, order=0, oom_score_adj=0

Before going to stock, I'm going to try to stop Home Assistant polling on the router.

What I don't get is it did it during the night, when there is no one using Internet and all devices are idle.

gnuton, 2023-10-27:
the  other time the OOM was triggered by aews this time by dnsmaq.

The problem is: you are running out of memory. 
Some apps needs to allocate more memory for their needs and the kernel cannot provide any free memory, 
This triggers the OOM.
I do not see this a bug or nothing I will be able to fix.

megosugit, 2023-10-27:
unless there is a memory leak.
Could be a result of repeated polling from Home Assistant, maybe the router fills up the memory and doesn't free it.

gnuton, 2023-10-28:
@megosugit in which program? from your 2 logs, there 2 different applications crashing, which makes clear that the problem is unlikely  in any of them.
Moreover the code runing on this router is the same which runs on all other routers and on stock firmware. If there is a memory leak that would affect everybody and not just you. 
Try to keep memory usage as low as possible. after all these routers have with limited  memory so threat them as such. 
If you need more router enable swap.

HerrBeef, 2023-10-28:
hi, i think i have a similar or the same problem.
2.4ghz keeps on crashing, every few hours.

This is the log:
Oct 28 20:10:21 kernel: wl0: PSM microcode watchdog fired at 90186 (seconds). Resetting.
Oct 28 20:10:21 kernel: wl0: reason = psm watchdog at 90186 seconds. corerev 42 ucode revision 1570.181 features 0x0426
Oct 28 20:10:21 kernel: psmdebug 0x00fe944c phydebug 0x41 macctl 0x84160403 maccmd 0x14
Oct 28 20:10:21 kernel: psm_brc 0x0600 psm_brc_1 0x0000 M_UCODE_DBGST 0x2
Oct 28 20:10:21 kernel: brwk_0 0x1018 brwk_1 0x0000 brwk_2 0x0430 brwk_3 0x7101 brwk_4 0x0000
Oct 28 20:10:21 kernel: ifsstat      0x18a0 ifsstat1     0xe0   txectl       0x481d txestat      0x401  
Oct 28 20:10:21 kernel: rxestat1     0x4110 rxestat2     0x40   rcv_frmcnt   0x0    rxe_rxcnt    0x5f   
Oct 28 20:10:21 kernel: wepctl       0x50   
Oct 28 20:10:21 kernel: aqmfifordy   0x2    wepstat      0x0    wep_ivloc    0x18   wep_psdulen  0x62   
Oct 28 20:10:21 kernel: PC :
Oct 28 20:10:21 kernel: R0: 0xff8224   0xc998e1   0xfe8004   0xff80ec  
Oct 28 20:10:21 kernel: R0: 0xff80da   0xc98004   0xec80e6   0xfe822d  
Oct 28 20:10:21 kernel: sched: RT throttling activated
Oct 28 20:10:21 kernel: R0: 0xc9937f   0xff80c4   0xff80dc   0xff801a  
Oct 28 20:10:21 kernel: R0: 0xec822d   0xff80c5   0xfe944c   0xff98e8  
Oct 28 20:10:21 kernel: R0: 0xec821e   0xfe98e1   0xec8007   0xff8117  
Oct 28 20:10:21 kernel: R0: 0xff80db   0xec8004   0xfe80e6   0xff80cb  
Oct 28 20:10:21 kernel: R0: 0xff80c7   0xff945b   0xfe93be   0xfe811a  
Oct 28 20:10:21 kernel: R0: 0xff8016   0xfe9440   0xff80d1   0xff8235  
Oct 28 20:10:21 kernel: R0: 0xff80d8   0xff8116   0xff80ce   0xc9822d  
Oct 28 20:10:21 kernel: R0: 0xc9822d   0xfe80de   0xec801c   0xc9821e  
Oct 28 20:10:21 kernel: R0: 0xc9822d   0xfe80de   0xec801c   0xc9821e  
Oct 28 20:10:21 kernel: R0: 0xff80cf   0xec822d   0xfe80de   0xfe944c  
Oct 28 20:10:21 kernel: R0: 0xff80ec   0xff80d0   0xfe822d   0xff80e2  
Oct 28 20:10:21 kernel: R0: 0xff80c9   0xff8227   0xfe9470   0xec8017  
Oct 28 20:10:21 kernel: R0: 0xff95da   0xff98e8   0xff80db   0xff80d5  
Oct 28 20:10:21 kernel: R0: 0xfe8224   0xfe821e   0xff821d   0xff8119  
Oct 28 20:10:21 kernel: phydebug :
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: 0x1        0x1        0x1        0x1       
Oct 28 20:10:21 kernel: pktproc      0x101  pktproc      0x101  pktproc      0x101  pktproc      0x101  
Oct 28 20:10:21 kernel: psm stack_status : 0x10
Oct 28 20:10:21 kernel: psm stack_entries:
Oct 28 20:10:21 kernel: 0x0226
Oct 28 20:10:21 kernel: 0x1470
Oct 28 20:10:21 kernel: 0x1631
Oct 28 20:10:21 kernel: 0x1cbf
Oct 28 20:10:21 kernel: 0x0ddb
Oct 28 20:10:21 kernel: 0x1ff9
Oct 28 20:10:21 kernel: 0x137d
Oct 28 20:10:21 kernel: 0x1fcc
Oct 28 20:10:21 kernel: wl0: fatal error, reinitializing, total count of reinit's[3]

I came from the latest AX92U Asus Software where it worked fine. i resetted the router and installed this, and since then i have this problem.

gnuton, 2023-10-29:
Oct 28 20:10:21 kernel: phydebug  these are debug messages

gnuton, 2023-10-29:
As for this issue there is nothing i can do here.
This is an OOM and the crashlog is an OOM too see lines below.
The reason for this is low memory on the router maybe due to mem intensive settings or applications with lack of swap.

```
0>Call trace:
...
May 5 07:05:06 crashlog: <4>[] __out_of_memory.isra.6+0xa4/0x148
May 5 07:05:06 crashlog: <4>[] out_of_memory+0x48/0x68
```

closed 2023-10-29 by gnuton

HerrBeef, 2023-10-29:
Okay then I misunderstood that.
But the 2.4g is restarting every few hours. The 5g is completely stable, eventhough it has way more devices connected to it. Is there any way to fix this or should I switch back to the original software.

gnuton, 2023-10-29:
@HerrBeef the 5ghz driver has been built without debug messages so you do not see the errors.

HerrBeef, 2023-10-29:
It is more stable, because my Smarthome devices run sadly on 2.4ghz only because these are cheap tuya wifi things, and these don't work right now really because of the crashes. On the other hand on my laptop and Google home's I use 5ghz wifi and had no disconnects yesterday at all, while the 2.4ghz devices regularly go offline, and 15m later online again. But I understand that the router can handle the error, but why are these errors happening at all, regular wifi 2.4g drop outs are not really normal.

gnuton, 2023-10-29:
@HerrBeef just to be clear, we are talking about the same drivers.  The wireless drivers are closed source components which are prebuilt by Asus for Merlin and theortically (despite debug message are on) they should just the same.
The problem is most likely being the way you have configured the driver
On google you may find more
https://www.snbforums.com/threads/kernel-wl0-psm-microcode-watchdog-fired-at-2222711-seconds-resetting.75247/

HerrBeef, 2023-10-29:
i found a "temporary" solution for my problem (even though i don't really understand how this problem even comes to be), i installed "RT-AX92U_388.2_2_0-gnuton1" on the router and everything works fine again. no random 2.4g wifi drops etc. saw that on the first post it said it was affecting 388.2 but i had "RT-AX92U_3004_388.4_0-gnuton1" installed. 

I also found that the official asus firmware "3.0.0.4_388_23630" has this problem, even though i reseted the router and didn't touch the default wifi settings except changing ssid and password. so something in the newest version makes the 2.4g wifi for my RT-AX92U not work reliable, why and how i dont understand.

CPngN, 2023-10-30:
> tuya

I have lots of "Tuya" smart devices also, and have also noticed this issue. 388.2 gnuton helped, but after several weeks, months perhaps, my mesh node's LED turned solid blue and it started having a hard time holding onto devices.  After a reboot most would reconnected, but within hours all but maybe 8 had dropped off.  

I tried to add my old RT-AC5300 as a node and it also "dies" after a while and I can't get into admin and nothing seems to be connected.  Unfortuantely it's too old to run the 388 codebase. But still, this whole AI Mesh experience has been far from awesome for me and my 60 devices.

megosugit, 2023-10-30:
> @megosugit in which program? from your 2 logs, there 2 different applications crashing, which makes clear that the problem is unlikely in any of them. Moreover the code runing on this router is the [...]

Home Assistant is a Home automation software running on another machine. From what I can see in Asus logs, it received ssh polling from Home Assistant around every 60s. I wonder if something in Asus FW is not able to free up the memory properly after each polling.
I have disabled it, we will see, for now uptime is: 3 days 6 hour(s) 17 minute(s) 40 seconds (🤞 )

> Try to keep memory usage as low as possible. after all these routers have small RAM modules so threat them as such. If you need more RAM please enable swap.

I remember seeing the memory usage was really high out of the box, the first time I turned it on.
I did a reset to factory initially, but even without any configuration, the memory usage is high (above 90% if I remember correctly).
Currently, it says 94%:
Used / Free / Total
480 MB / 32 MB / 512 MB

Do you think I should bother with adding an usb drive for swap ? I wished I could keep it simple.

gnuton, 2023-11-08:
1. were you closing your SSH connections?
2. no that I know and the answer is not easy. There may be something close to your router (AP or other radio devices) which interfeer with your router, and makes the radio to  reinitialize. There is no description of why this happens, so I cannot be sure of what's the reason. This happens only on the 40Mhz bandwidth. One reason for this interference could be an AP which is using a nearby channel on the 2.4 GHz band for instance.

megosugit, 2023-11-08:
1. I don't have control on this option, but it should close it as soon as it gets the data.
Is there a way to see if ssh connections are accumulating on the router ?

2. Unfortunately, there are a lot of other APs around, maybe the router doesn't like all this noise.
If it is stable as it is, I'm happy keeping the 20Mhz. It is mostly for domotic devices, they don't need a lot of bandwidth.

3. About the wl0 fatal error that keep happening, I found a lot of other people online having this issue, but no solution, except some mention about defective router.
Are you able to understand the reason given in the logs ?

> Nov  8 09:22:40 kernel: wl0: PSM microcode watchdog fired at 1047210 (seconds). Resetting.
> Nov  8 09:22:40 kernel: wl0: reason = psm watchdog at 1047210 seconds. corerev 42 ucode revision 1570.181 features 0x0426
> Nov  8 09:22:40 kernel: psmdebug 0x00ff8014 phydebug 0x41 macctl 0x84160403 maccmd 0x4
> Nov  8 09:22:40 kernel: psm_brc 0x0600 psm_brc_1 0x0080 M_UCODE_DBGST 0x2
> Nov  8 09:22:40 kernel: brwk_0 0x049c brwk_1 0x0002 brwk_2 0x0430 brwk_3 0x7123 brwk_4 0x0000
> Nov  8 09:22:40 kernel: ifsstat      0x9ea0 ifsstat1     0xf0   txectl       0x4000 txestat      0x401  
> Nov  8 09:22:40 kernel: rxestat1     0x4009 rxestat2     0x2b40 rcv_frmcnt   0x0    rxe_rxcnt    0x11a  
> Nov  8 09:22:40 kernel: wepctl       0x50   
> Nov  8 09:22:40 kernel: aqmfifordy   0x10   wepstat      0x0    wep_ivloc    0x1e   wep_psdulen  0x1e   
> Nov  8 09:22:40 kernel: PC :
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: R0: 0xff8014   0xff8014   0xff8014   0xff8014  
> Nov  8 09:22:40 kernel: phydebug :
> Nov  8 09:22:40 kernel: 0x41       0x41       0x41       0x41      
> Nov  8 09:22:40 kernel: 0x41       0x41       0x41       0x41      
> Nov  8 09:22:40 kernel: 0x41       0x41       0x41       0x41      
> Nov  8 09:22:40 kernel: 0x41       0x41       0x41       0x41      
> Nov  8 09:22:40 kernel: 0x40       0x40       0x40       0x40      
> Nov  8 09:22:40 kernel: 0x40       0x40       0x40       0x40      
> Nov  8 09:22:40 kernel: 0x0        0x0        0x0        0x0       
> Nov  8 09:22:40 kernel: 0x31       0x31       0x31       0x31      
> Nov  8 09:22:40 kernel: pktproc      0x101  pktproc      0x101  pktproc      0x101  pktproc      0x101  
> Nov  8 09:22:40 kernel: psm stack_status : 0x10
> Nov  8 09:22:40 kernel: psm stack_entries:
> Nov  8 09:22:40 kernel: 0x0c27
> Nov  8 09:22:40 kernel: 0x1470
> Nov  8 09:22:40 kernel: 0x11d3
> Nov  8 09:22:40 kernel: 0x11ce
> Nov  8 09:22:40 kernel: 0x0d7d
> Nov  8 09:22:40 kernel: 0x1e1f
> Nov  8 09:22:40 kernel: 0x1d56
> Nov  8 09:22:40 kernel: 0x1fff
> Nov  8 09:22:40 kernel: wl0: fatal error, reinitializing, total count of reinit's[2916]

gnuton, 2023-11-08:
1. you can ps -ax and grep for sshd or I guess you can configure ssh to accept a max num of concurrent connections.
2. isn't the wl0 fatal connection related to fact your router cannot use the 40mhz bandwidth because of the noise around? Those are the debug messages for this

megosugit, 2023-11-08:
1. would you know where to look for configuring ssh max concurrent connections ? I had a look without finding anything.
If not possible, I could try to lower the idle timeout to 1 or 2 minutes.

2. the wl0 fatal connection are still happening even though I have disabled the 20/40Mhz for several days.
Why would it reset the wl0 interface if 40Mhz is not used ?
Are we sure wl0 means 2.4Ghz radio ?

megosugit, 2023-12-08:
just to report that after disabling external ssh polling, the router random reboots are not happening anymore.
And changing 2.4Ghz Channel bandwidth wifi settings from 20/40 Mhz to 20Mhz helped a lot. Devices are not dropping from wifi anymore.