gnuton/asuswrt-merlin.ng #479: Firmware update check throws error, error also Asus specific.

Yakumounr, 2023-10-08:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
3004.388.4_0-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Firmware update check is failing currently, and the error message for it states:

 "The router cannot connect to ASUS server to check for the firmware update. Please download the latest firmware from ASUS Product Support Site and do manually upgrade. Check on [FAQ](http://192.168.1.1/Advanced_FirmwareUpgrade_Content.asp) for more instructions." 
 
 which should not be the case for Gnuton firmware.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration'
2. Click on 'Firmware Upgrade'
3. Click on 'Check'
4. See error

**Expected behavior**
Update check should connect to appropriate service and report update/same ver/or Gnuton server error as apropriate.

gnuton, 2023-10-24:
The new format is 

RT-AX58U#FW3004.388.4#EXT0

gnuton, 2023-10-26:
Fixed in the stable

closed 2023-10-26 by gnuton