gnuton/asuswrt-merlin.ng #480: Firmware Upgrade page Current Version details change seconds after page load.

Yakumounr, 2023-10-08:
**Router Model Affected**
Models:  DSL-AX82U

**Firmware Version Affected**
3004.388.4_0-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
A weird quirk on the Firmware Upgrade page, Current Version shown loads initially as "3004.388.4_0-gnuton0_beta2", approximately four seconds afterwards with no user interaction needed it pops into "3004.388.4-gnuton0_beta2" dropping the "_0" for some reason.
Tested on Chrome and Firefox.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration' from the left menu
2. Click on 'Firmware Upgrade' tab.
4. Note the Current Version displayed
5. Wait 4+ seconds, note any change

**Expected behavior**
Firmware version information should stay consistent.

Katana1074, 2023-10-18:
I can confirm this bug......

Router Model Affected
Models: DSL-AX82U

Firmware Version Affected
3004.388.4_0-gnuton0_beta2

gnuton, 2023-10-26:
Fixed in the stable

Yakumounr, 2023-10-27:
It still happens in the stable release
"3004.388.4_0-gnuton1" is changing to "3004.388.4-gnuton1"