gnuton/asuswrt-merlin.ng #483: speed drops suddently for killer wifi card

bobyang3, 2023-10-24:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
388.2_2_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
don't know

**Describe the bug**
[image]

Hello,
I have encountered a peculiar problem. Suddenly, the speed of my internet connection drops significantly, with the Tx and Rx dropping to 14.4Mbps and 1Mbps respectively, from the previous speeds of 866.7Mbps. Unfortunately, the speed does not recover on its own. The only solution I have found is to disconnect and reconnect to the Wi-Fi network or reboot the router.

This issue also occurs sometimes (approximately 30% of the time) when I wake up laptops in the morning. The router's Network Map shows a speed of only 14.4Mbps and 1Mbps in these cases. However, when I check the Wi-Fi status using Powershell, it indicates that the connection is functioning properly.

Everything was working fine with version 386, but the problems started after I upgraded to version 388.2.2. This issue seems to be specific to laptops with Killer Wi-Fi cards, as these laptops do not experience any issues with other routers.

Here is some additional information about my setup:
1. I have scheduled a reboot at 3am every Monday, Wednesday, and Friday.
2. I have connected a hard disk to the USB port on the router and set up file sharing.
3. I have an OpenVPN server set up, but it is currently disabled.
4. I have tried 4 different killer drivers with the same result.
5. I am in the same location all the time because those laptops are used as a desktop

I hope this rephrased explanation helps. Let me know if there's anything else I can assist you with.

**To Reproduce**
unknow.

**Expected behavior**
speed won't drop

**Screenshots**
[image]
[image]

gnuton, 2023-10-26:
can you please run a https://www.speedtest.net when you get a WiFi drop? 
Just to be sure what you read there it's real. 
Moreover if the other Wifi cards work fine, and the killer show this problem I would rather think the problem is related to drivers of these network adapters.

bobyang3, 2023-10-26:
I ran the speedtest.net and google speed test with the same result of slowness.

yes. agreed about the driver but it is so odd because it was working well in 386.xxx (within this machine) and other not Asus routers. It seems like the problem only exists for 388 firmware.

Thank you !