gnuton/asuswrt-merlin.ng #484: Redirect webui access to www.asusrouter.com

doggi87, 2023-10-28:
**Router Model Affected**
Models: TUF AX5400

**Firmware Version Affected**
3004.388.4_0-gnuton1_tuf

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
In Administration -> System
"Redirect webui access to www.asusrouter.com" is No per default but still the router goes to www.asusrouter.com when accessing the router from example 192.168.1.1
When turning it to Yes or No it't not using the setting.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Administration
2. Click on System
3. Scroll down to "Redirect webui access to www.asusrouter.com"
4. Turn it to Yes or No both are the same for this setting

argh0815, 2023-10-29:
The condition marked in bold is missing in Main_Login.asp of the TUF (and ROG) version:
**if ('<% nvram_get("http_dut_redir"); %>' == '1') {**
var isRouterMode = ('<% nvram_get("sw_mode"); %>' == '1') ? true : false;

var header_info = [<% get_header_info(); %>][0];
var ROUTERHOSTNAME = '<% nvram_get("local_domain"); %>';
var domainNameUrl = header_info.protocol+"://"+ROUTERHOSTNAME+":"+header_info.port;
var chdom = function(){window.location.href=domainNameUrl};
(function(){
	if(ROUTERHOSTNAME !== header_info.host && ROUTERHOSTNAME != "" && isRouterMode){
		setTimeout(function(){
			var s=document.createElement("script");s.type="text/javascript";s.src=domainNameUrl+"/chdom.json?hostname="+header_info.host;var h=document.getElementsByTagName("script")[0];h.parentNode.insertBefore(s,h);
		}, 1);
	}
})();
**}**

You can work around that by using your own version and mount bind it in services-start or wait for a fix or use the Merlin theme.

gnuton, 2023-11-16:
fixed by 42bf052cc3